RoganDawes/P4wnP1_aloa #44: keymap latam

vay3t, 2018-11-28:
is possible p4np1 support for latam keymap?
i have in rubberducky format https://gist.github.com/vay3t/ec99702ee9f0554a35c9763050892400

vay3t, 2018-12-07:
I have not tested it but for Latin it should be like that

```
{
	"Name": "ES",
	"Description": "Contains Spanish key mappings",
	"Mapping": {

            "CTRL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "CONTROL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "LEFT_CTRL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "RIGHT_CTRL": [{"Modifiers": ["MOD_RIGHT_CONTROL"], "Keys": []}],
            "ALT": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": []}],
            "LEFT_ALT": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": []}],
            "RIGHT_ALT": [{"Modifiers": ["MOD_RIGHT_ALT"], "Keys": []}],
            "SHIFT": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": []}],
            "LEFT_SHIFT": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": []}],
            "RIGHT_SHIFT": [{"Modifiers": ["MOD_RIGHT_SHIFT"], "Keys": []}],
            "GUI": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "WIN": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "LEFT_GUI": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "RIGHT_GUI": [{"Modifiers": ["MOD_RIGHT_GUI"], "Keys": []}],

            "ESC": [{"Modifiers": [], "Keys": ["KEY_ESC"]}],
            "ESCAPE": [{"Modifiers": [], "Keys": ["KEY_ESC"]}],

            "F1": [{"Modifiers": [], "Keys": ["KEY_F1"]}],
            "F2": [{"Modifiers": [], "Keys": ["KEY_F2"]}],
            "F3": [{"Modifiers": [], "Keys": ["KEY_F3"]}],
            "F4": [{"Modifiers": [], "Keys": ["KEY_F4"]}],
            "F5": [{"Modifiers": [], "Keys": ["KEY_F5"]}],
            "F6": [{"Modifiers": [], "Keys": ["KEY_F6"]}],
            "F7": [{"Modifiers": [], "Keys": ["KEY_F7"]}],
            "F8": [{"Modifiers": [], "Keys": ["KEY_F8"]}],
            "F9": [{"Modifiers": [], "Keys": ["KEY_F9"]}],
            "F10": [{"Modifiers": [], "Keys": ["KEY_F10"]}],
            "F11": [{"Modifiers": [], "Keys": ["KEY_F11"]}],
            "F12": [{"Modifiers": [], "Keys": ["KEY_F12"]}],

            "SYSRQ": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "PRINT": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "PRINTSCR": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "SCROLLLOCK": [{"Modifiers": [], "Keys": ["KEY_SCROLLLOCK"]}],
            "SCROLL": [{"Modifiers": [], "Keys": ["KEY_SCROLLLOCK"]}],
            "PAUSE": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],
            "BREAK": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],
            "BRK": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],

            "|": [{"Modifiers": [], "Keys": ["KEY_GRAVE"]}],
            "1": [{"Modifiers": [],	"Keys": ["KEY_1"]}],
            "2": [{"Modifiers": [],	"Keys": ["KEY_2"]}],
            "3": [{"Modifiers": [],	"Keys": ["KEY_3"]}],
            "4": [{"Modifiers": [],	"Keys": ["KEY_4"]}],
            "5": [{"Modifiers": [],	"Keys": ["KEY_5"]}],
            "6": [{"Modifiers": [],	"Keys": ["KEY_6"]}],
            "7": [{"Modifiers": [],	"Keys": ["KEY_7"]}],
            "8": [{"Modifiers": [],	"Keys": ["KEY_8"]}],
            "9": [{"Modifiers": [],	"Keys": ["KEY_9"]}],
            "0": [{"Modifiers": [],	"Keys": ["KEY_0"]}],
            "'": [{"Modifiers": [],	"Keys": ["KEY_MINUS"]}],
            "¿": [{"Modifiers": [], "Keys": ["KEY_EQUAL"]}],

            "°": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_GRAVE"]}],
            "!": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_1"]}],
            "\"": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_2"]}],
            "#": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_3"]}],
            "$": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_4"]}],
            "%": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_5"]}],
            "&": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_6"]}],
            "/": [{"Modifiers": ["MOD_LEFT_SHIFT"],	"Keys": ["KEY_7"]}],
[... 151 lines cut ...]
            "RIGHT_ARROW": [{"Modifiers": [], "Keys": ["KEY_RIGHT"]}],
            "LEFT_ARROW": [{"Modifiers": [], "Keys": ["KEY_LEFT"]}],
            "DOWN_ARROW": [{"Modifiers": [], "Keys": ["KEY_DOWN"]}],
            "UP_ARROW": [{"Modifiers": [], "Keys": ["KEY_UP"]}],

            "COMPOSE": [{"Modifiers": [], "Keys": ["KEY_COMPOSE"]}],

            "NUMLOCK": [{"Modifiers": [], "Keys": ["KEY_NUMLOCK"]}],
            "NUM": [{"Modifiers": [], "Keys": ["KEY_NUMLOCK"]}],

            "F13": [{"Modifiers": [], "Keys": ["KEY_F13"]}],
            "F14": [{"Modifiers": [], "Keys": ["KEY_F14"]}],
            "F15": [{"Modifiers": [], "Keys": ["KEY_F15"]}],
            "F16": [{"Modifiers": [], "Keys": ["KEY_F16"]}],
            "F17": [{"Modifiers": [], "Keys": ["KEY_F17"]}],
            "F18": [{"Modifiers": [], "Keys": ["KEY_F18"]}],
            "F19": [{"Modifiers": [], "Keys": ["KEY_F19"]}],
            "F20": [{"Modifiers": [], "Keys": ["KEY_F20"]}],
            "F21": [{"Modifiers": [], "Keys": ["KEY_F21"]}],
            "F22": [{"Modifiers": [], "Keys": ["KEY_F22"]}],
            "F23": [{"Modifiers": [], "Keys": ["KEY_F23"]}],
            "F24": [{"Modifiers": [], "Keys": ["KEY_F24"]}]
    }
}
```

mame82, 2018-12-08:
From first inspection this looks legit, could you test it?
If you add the proper file with content from above, it should be usable if you call `layout("latam")` from a HIDScript.

I'd happily accept a PR in case it works to include it in the next version

vay3t, 2018-12-15:
Today I was doing some tests and the hid attack did not work. I'm using the release. If you need something, some log you can ask me

himuura, 2019-07-02:
tested it just know, the latin keyboard described above. Doesnt work.

vay3t, 2019-07-10:
I'll fix it shortly. do you have twitter or telegram? @himuura

himuura, 2019-07-10:
unfortunetely no mate, why?

vay3t, 2019-07-10:
to be able to have better feedback, in terms of inconveniences with the keymap

himuura, 2019-07-10:
i could create one if it helps :)

himuura, 2019-07-12:
done mate, @shadow_ownz