RoganDawes/P4wnP1_aloa #49: Android ADB host and keyboard possible?

pabloko, 2018-12-12:
Hello, Im on the need of running some adb commands on a lot of android devices, in order to get adb to work, a keyboard could be used to enable it.

Its possible to have both?

regards!

mame82, 2018-12-12:
If I understand you correctly, you want to interface with ADB over USB. This is technically possible with an USB OTG adapter on Pi's end. But it wouldn't be possible to run a keyboard at the same time.

Although the DesignWare USB controller supports dual role mode, it can't act as host and device at the same time.

The dual role mode driver used by P4wnP1 toggles into host mode, as soon as an OTG adapter is attached (ID pin shorted). Software-wise the dwc2 USB driver uses a completly different driver stack from this point (in fact some registers of the UDC chip couldn't be accessed when in host mode).

Short answer is: No.
Nice answer is: Target ADB over Ethernet and you're good to go (WiFi or Bluetooth, not USB ethernet)

pabloko, 2018-12-12:
Well theres no way to forward adb to ip if its not proxified over usb first...

Is there a way where i could just toggle the keyboard when adb is active on the android system, then change the usb as normal device for adb? (as i have a previously used adb fingerprint keys it should not ask to allow them)

Many thanks, and tbh amazing project!

mame82, 2018-12-12:
Was referring to "Wireless ADB" which exists since Android 4.0. 

No need for USB, availability depends on device

pabloko, 2018-12-12:
It doesnt seem to be an option for me, it seems i need usb no matter what... thats why ive asked if theres some toggle possible.