RoganDawes/P4wnP1_aloa #52: Press printscreen in HID?

aidan-walden, 2018-12-17:
Ive tried both press("PRINTSCREEN") and press("PRNTSCRN") and neither seem to do anything. Is there a way to view all available keys that are valid inputs to the press function?

Swiftb0y, 2018-12-17:
Well, the press function will call [StringToPressKeycombo](https://github.com/mame82/P4wnP1_aloa/blob/master/hid/keyboard.go#L205) internally which will eventually call [mapKeyNameToReport](https://github.com/mame82/P4wnP1_aloa/blob/master/hid/keyboard.go#L296) which will eventually look for the keysequence in the loaded mapping. However since printscreen is not a character, it won't find anything in there and will look into keyboard_globals.go and find it there [KEY_SYSRQ](https://github.com/mame82/P4wnP1_aloa/blob/master/hid/keyboard_globals.go#L688)

### TLDR:
Try `press("KEY_SYSRQ")` or something similar.

mame82, 2018-12-17:
KEY_SYSREQ is mapped to

- `PRINT`
- and `PRINTSCR`

closed 2018-12-28 by mame82