RoganDawes/P4wnP1_aloa #54: Some keys are wrong

billym357, 2018-12-28:
y and z are mixed up. If you type "y" it'll show up as "z" on the attacked pc and if you type "z" it'll show up as "y".        Also  '  shows up as   |

mame82, 2018-12-28:
Could you give details on the language layout used by P4wnP1 and the language layout used by the USB host

mame82, 2018-12-29:
If you don't set the language layout in your HIDScript with `layout("us")` the layout from the last call to `layout` is used.

If `layout` was never called from a HIDScript, the boot setting is used, which defaults to EN_US.

It seems you're using a layout, which doesn't match your target.

If the target host uses EN_US and you use `layout("de")` on P4wnP1 thus would lead to the described behavior (z and y swapped).

Please assure to use the proper layout (see section 2.2 of the readme).

US layout is tested and confirmed working

mame82, 2018-12-30:
The files are stored in images which reside in under `/usr/local/P4wnP1/ums/flashdrive` for flashdrive and under `/usr/local/P4wnP1/ums/cdrom` for ISOs.

The tool in `/usr/local/P4wnP1/helper/genimage` could be used to create new ISO or flashdrive images (from existing folders if needed)

mame82, 2019-07-24:
Note: Include statements on mass storage image locations in FAQ: "Where are the images for USB Mass Storage placed?"