RoganDawes/P4wnP1_aloa #56: How can we connect external Status LED

itsStar, 2018-12-31:
I am using [This BAD USB Connector ](https://ae01.alicdn.com/kf/HTB1dZ3vXcrrK1RjSspaq6AREXXa2.jpg) And it connect the P4wnP1 Upside-down with the target and i am unable to see the status light.
Thats why i want to connect external 2 color LED for status updates while using headless during HID attacks.

Can you guide me which GPIO pins can be used for this process.
Thanks in advance :)

itsStar, 2018-12-31:
I have found this [Video](https://www.youtube.com/watch?v=_REhMZJ8GJY)
Explaining the scenario on Add An External Sd Activity Light Raspberry Pi 3 Retropie Or Raspbian
Will it work on P4wnP1_aloa ?

itsStar, 2019-01-02:
Is there any HID based status mechanism in P4wnP1 ?

mame82, 2019-01-03:
HIDScript functionality is currently seperated from GPIO and TriggerAction functionality.

Anyways, bashscripts could be used to glue everything together. Based on the return values of HIDScripts started with the CLI client from a bashscript, almost all functionality could be used, including GPIO and LED control

itsStar, 2019-01-06:
Till now i have attached an RGB LED to GPIO Pin  13, 19, 26 and wrote a python script to turn them on OR off. This is how it looks like now 

[image]

Now i want to find a way by which i could turn them ON OR Off during an HID script. Can you give me some clue which files i should look for this purpose OR is there any suggestion you may give me which help me in this regard.

itsStar, 2019-01-06:
To be more precise can you tell me how can i call a local script (python on P4wnP1) from HID script to run my python code to turn on OR off the LED

netsecninja, 2019-01-06:
@itsStar, The way P4wnPi is set up now, HID scripts are just that, HID input scripts. You cannot execute any other P4wnPi actions from within. You can use a trigger action to run a bash script, which calls your python code. You can set several different actions for the same trigger to get things running at the same time (examples on the readme.md). So a service started trigger to change the LED, then the another service started trigger to run the HID script. But that's the end of the chain, as there is no automated feedback from the P4wnPi or victim that the code has run to begin another action. There are group channels, but I'm not seeing a way to use them in this situation, or maybe I don't understand them well enough.

Example: You are wanting a red LED when the script is starting, then a blue when it's running the attack, then a green when it's complete. Unless I'm missing something, I'm only aware of one way of doing it. Learn how to interact with P4wnP1 from command line (P4wnP1_cli), you can create a single bash script that will monitor for trigger actions, execute HID scripts, run python code to change LEDs, and many other things.

Unfortunately, this means that we are back to a single automation script, which was the old method of doing things. It does present a limitation that hopefully @mame82 can overcome, as the web UI is an awesome addition, and I'd like to do everything through it. If I'm wrong, please feel free to correct me. I'm still learning to use this tool. Also realize it's still in Alpha, so keep the questions and comments coming.

itsStar, 2019-01-06:
@netsecninja  As i understand HID script is only a file which is being parsed by a master script and that master Script then parse it and execute it. 
SO it could be so simple as passing a new method LED(color) to Master script of P4wnP1 like other functions (press(), OR type()) are parsed and being executed by P4wnP1 
then it can be merged with the python function to turn on OR off the LEDs

Actually this is important in situation where we just want to attach the P4wnP1 to target computer where it boots up and do its work covertly without any user interaction, so feedback from P4wnP1 is important.

Here is an Example of what i have in Mind

layout('us');			// US keyboard layout
typingSpeed(0,0)	// Wait 100ms between key strokes + an additional random value between 0ms and 150ms (natural)
**LED(B);**
press("GUI r");
delay(500);
type("powershell\n")
delay(100);
type("powershell -w hidden -executionpolicy bypass -c ((gwmi win32_volume -f 'label=''RPIW''').Name+'WP.ps1')")
delay(100);
press("ENTER");
delay(1000);
**LED(G);**

I hope this is not that hard

netsecninja, 2019-01-06:
Except as documented the web UI is the only method to create/modify Master templates at this time.

> Using the webclient is a convenient way to define Master Templates, as it supports you by only allowing to select templates, which have been already stored for the respective sub systems (and currently the webclient is the only way to define Master Templates).

itsStar, 2019-01-06:
@netsecninja Actually i am talking about the Master script (P4wnP1_aloa-masterCode>hid>controller.go) not Master Template.

controller.go is the script where all parsing is being done. And maybe there are other locations too.

mame82, 2019-01-08:
Statement on MasterScript (go file) is wrong, the file is part of the source code for the service binary.

I'm currently on the road and thus have to answer this questions later, as I assume others are interested, too.

For now in short words: The HIDScript language is interpreted by a modified JavaScript engine (not parsed, like f.e. static DuckyScript).
The JavaScript engine knows nothing about P4wnP1 and thus couldn't control its workflow (yet). I gonna explain the reasons later. In contrast to this, P4wnP1 could influence the HIDScript engine in multiple ways, another thing I'll explain, as soon as I have time to.

itsStar, 2019-01-08:
@mame82 Thanks for the quick  reply have a safe journey

itsStar, 2020-02-07:
@mame82 Hello guys is there a chance to add this functionality in near future :) please share your thoughts in it Thanks