RoganDawes/P4wnP1_aloa #59: GB Layout isn't working

Zetascrub, 2019-01-09:
I'm trying to create a HID script, here's a sample

`layout('GB');           // GB keyboard layout
typingSpeed(100,150)    // Wait 100ms between key strokes + an additional random value between 0ms and 150ms (natural)

press("WIN r");
type("Powershell")`

However, "e" is actually typed as "é", I've checked the layout

If I select the layout as "us" then it'll type "e" just fine.

I've tried changing some of the keymapping but I couldn't figure it out :\

Otherwise, awesome upgrade.

Zetascrub, 2019-01-09:
Example

[image]

craigsblackie, 2019-01-26:
Try removing the line:
"e": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_E"]}]

Zetascrub, 2019-01-30:
> Try removing the line:
> "e": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_E"]}]

I've tried this, but it hasn't worked.

HazzaOnYT, 2019-03-26:
[image]
same problem here :)

HazzaOnYT, 2019-03-26:
Also N is missing.

datak0w, 2019-06-02:
Any updates on this? Same problem here, GB. Other characters displayed incorrectly, e.g. “  and ”. 
"n" is missing, "e" is displayed as "é" , and so on. 
Is it the encoding, maybe?

mame82, 2019-06-02:
Seems I have multiple errors in GB layout. They are caused by doubled character assignments in the layout files. The following PR fixed `e` : https://github.com/mame82/P4wnP1_aloa/pull/71/files

I have to rework the whole GB map file and tag it with next minor release.

PRs welcome

Zetascrub, 2019-06-07:
Currently trying to debug this, interesting that I removed all keymaps, rebooted and HID still works and if I swap between GB and US, it still does the same issue as before

turkishlitecoin, 2019-06-25:
Having an issue with the pipe command (|). 
When entered on HIDscript, the ~ the displayed.

Initially thought it was the OS what the attack/script was running against, but this occurs on various Windows releases (7&10), as well as Linux distributions.

I have looked at the script in the HIDscript folder on the Pi, and can see that the pipe command is visible, so appears to be transformed between the script and the target machine/OS.
All are running uk/gb keyboard layouts.

evilbobbins, 2019-06-25:
A lot of the gb mapping it out and creates odd results, I have fixed some by updating the key map but some seem to default back to fr or us even when remapped, if this was cleaned up this would be an amazing platform way better than a bash bunny etc

Zetascrub, 2019-08-06:
Hi, any progress on this?

I know there's a million other things to sort out first, but I'm really eager to play around with this device for HID attacks, however the keyboard layout is preventing this :/

slipdip123, 2020-02-01:
Issue still seems prevalent on my p4wnp1 aloa

mame82, 2020-02-05:
fixed in current dev branch

lggarcia, 2020-06-23:
have the PIPE problem as well....

type("$LogicalDisks | ?{($_.Path).Path -eq ($PartitionToLogicalDisk.Dependent)}\n");

on this code, using ES (spain) keyboard layout, the PIPE simply desappears when typed ! 
also, how can I type " on the hidscript ?