RoganDawes/P4wnP1_aloa #63: can this work on zero?

etnaaflonne123, 2019-01-14:
ok I put this on pi zero (not zero w) cuz its what I got  I try to make it work via the usb and network shows up but I cannot connect to it and I have no other way to log in I see a log in screen if I use a monitor and it wont work via pi 2

when I use the old version and I try to crack a password (password) is the actual password 
I get the picture below can anybody tell me if its cracking or searching or what I let it sit there a very long time the screen sleep outs and nothing happens and I cannot get the new one working 

id like to start by doing windows unlock and actually using it via a business type thing ex: pc lock put in unlock something to leave for (a pc idiot)  when I cannot be there.

advice? or is this thing actually working I have no idea I cant find anything on google
or can you make the new one work via zero not zero w?

attached is what the pic of the screen on the old version (so is it working? )
[image]

etnaaflonne123, 2019-01-14:
this is the 2nd pic 
[image]

whackyhack, 2019-06-09:
Unfortunately P4wnP1 ALOA will not initiate USB Ethernet on a Pi Zero without "W" because it waits for output from the native WiFi chip. (See #119.)  I just went around and bought a pi0w, and continue to test legacy P4wnP1 on pi0. (For what you wanted to do, p4wnp1 should work.)