RoganDawes/P4wnP1_aloa #64: quick HOWTO for linux: USB networking + internet access for the p4wnp1 aloa

xorr0, 2019-01-15:
# quick HOWTO for linux: USB networking + internet access for the p4wnp1 aloa

- initially I was trying for USB serial on the p4wnp1 aloa, but that required one to manually enable the USB serial option (and disable one of the other USB options). 
- this method works out-of-the-box, and works better for me personally.

## after connecting the p4wnp1 aloa to your computer via USB, setup your linux host computer to forward IP packets
`$ echo "1" | sudo tee /proc/sys/net/ipv4/ip_forward`
   * if doing this via a script, add ` > /dev/null`:
   `$ echo "1" | sudo tee /proc/sys/net/ipv4/ip_forward > /dev/null`

`$ sudo iptables -A POSTROUTING -t nat -j MASQUERADE -s 172.16.0.0/30`
`$ sudo ifconfig usb0 172.16.0.2 netmask 255.255.255.252`

## test connectivity, ping the p4wnp1 aloa at 172.16.0.1, and you should get a successful response
`$ ping 172.16.0.1`

## ssh to the p4wnp1 aloa at 172.16.0.1 (DEFAULT SSH PASSWORD: `toor`)
`$ ssh root@172.16.0.1`

## setup a default route on the p4wnp1 aloa back to the linux host device:
`root@kali:~# route add default gw 172.16.0.2 usbeth`

## test connectivity via ping to the internet, you should get a successful response
```
root@kali:~# ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=57 time=15.0 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=57 time=17.3 ms
^C
--- 1.1.1.1 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 3ms
rtt min/avg/max/mdev = 15.021/16.177/17.333/1.156 ms
```
- _OPTIONAL_: feel free to add custom nameserver(s) to `/etc/resolv.conf`
`root@kali:~# echo "nameserver 1.1.1.1" >> /etc/resolv.conf`
```
root@kali:~# cat /etc/resolv.conf
nameserver 8.8.8.8
nameserver 1.1.1.1
```

xorr0, 2019-01-19:
* even more streamlined; bash functions (utilizes [ssh-pass](https://www.cyberciti.biz/faq/noninteractive-shell-script-ssh-password-provider/))
```
p4wnp1 () {
        echo "1" | sudo tee /proc/sys/net/ipv4/ip_forward > /dev/null
        sudo iptables -A POSTROUTING -t nat -j MASQUERADE -s 172.16.0.0/30
        sudo ifconfig usb0 172.16.0.2 netmask 255.255.255.252
        ping -c 1 172.16.0.1
        if [ $? -eq 0 ]
        then
                sshpass -p "toor" ssh -f root@172.16.0.1 "route add default gw 172.16.0.2 usbeth"
                echo "\n[+]  SUCCESS: access P4wnP1 via USB CDC Ethernet @ 172.16.0.1/30\n"
        else
                echo "\n[!]  FAILURE, something went wrong\n"
        fi
}

nop4wnp1 () {
        sshpass -p "toor" ssh -f root@172.16.0.1 "route del default gw 172.16.0.2 usbeth"
        sleep 2
        echo "0" | sudo tee /proc/sys/net/ipv4/ip_forward > /dev/null
        sudo iptables -t nat -F POSTROUTING
        if [ $? -eq 0 ]
        then
                echo "\n[+]  SUCCESS: P4wnP1 connectivity removed\n"
        else
                echo "\n[!]  FAILURE, something went wrong\n"
        fi
}
```

felixperez, 2019-03-13:
Wow, good joob! We can connect us P4wnP1 to internet and works like ever?
If us host is windows, what we have to do?

Thanks men :)

mame82, 2019-03-13:
For Windows:

1) Enable RNDIS on P4wnP1
2) Reconfigure `usbeth` to use DHCP client instead of DHCP server
3) On the Windows host enable Internet Connection Sharing (ICS) for the Network adapter which provides Internet access
4) Configure ICS to share the connection with the RNDIS adapter deployed by P4wnP1 (after attaching via USB)

felixperez, 2019-03-14:
[image]

Ok, i connect with my phone to 172.24.0.1, acced  the web interface, goes to "Networking", selected "usbeth" and changed DHCP-SERVER to DCHP-CLIENT, pressed "DEPLOY" but... 

In my W10 i can't find the network interface "Red no identificada" (spanish) --> "Unidentified network" (english).
With this error we can't ICS us real w10 connection (Wifi).

I have enabled the RNDIS "USB SETTINGS"--> RNDIS (by default)

poritz, 2019-06-15:
I have problems with bith solutions (linux on VM). 
My quetsion regards Windows: When I choose usbeth with DHPC_CLIENT on the web client and hit deploy, it sets back to bteth with DHPC_CLIENT. I've tried multiple variations to check if I could see whether it comes from some combination of settings, but I couldn't find anything. With this settings I can ssh into the P4wnP1 but can't bing the host.
With Raspian Stretch on zero I can access the internet so I assume it's not the firewall. I turned it down just to check and it does not help.
Is there another way to change to dhpc client on usbeth, maybe via console? Or can you think of workaround?

poritz, 2019-07-11:
Hi, I'm having a Problem with sshing into my p4wnp1 after getting dynamic ip (in my case 192.168.137.1). What i tried so far:
-sshing into 172.16/24.0.1 could resch it, meither via phone nor putty on targetmachine
-sshing into 192.168.137.1 access denied
Looking forward to hear some suggetsions :)

anduril-glitch, 2020-01-15:
> 
> 
> Hi, I'm having a Problem with sshing into my p4wnp1 after getting dynamic ip (in my case 192.168.137.1). What i tried without success so far:
> -sshing into 172.16/24.0.1 couldn't reach it, neither via phone nor putty from targetmachine
> -sshing into 192.168.137.1 connection refused (via putty and phone)
> -using the discribed method on a vm ware kali linux distro
> -using a ethernet to usb adapter directly plugged into my router and sshing into p4p1 via phone network unreachable/ temporary nameressolve error
> Looking forward to hear some suggetsions on what to try next :)
> 
> btw I also tried to install software (ettercap) manually but cmake is missing and installing cmake is not working due to make not working properly XD

I have the same problem.

Could you find a solution?

Thank you

A +

Zsouza7612, 2024-01-23:
ive ben tryying evything keeps denying my password  I can get it hooked up as rndis but won't work with my org so I can't uee my keyboard and can't ssh into cause passwd

Zsouza7612, 2024-01-23:
otg cable