RoganDawes/P4wnP1_aloa #73: Saving Files to SD Card

caretaker0815, 2019-01-27:
Is it possible to load files from a PC to the SD card of the Raspberry Pi when I am connected via USB? 
And can I transfer files from the SD card to the PC? Maybe via HIDScript?

itsStar, 2019-01-28:
Yes this is possible for this you can take a look @ wifigrabber HID Script 
Steps are as follows.

Find the Drive letter of your USB Drive via executing 
((gwmi win32_volume -f 'label=''RPIW''').Name')") in powershell

then you can easily copy and past the files to that location.

mame82, 2019-01-28:
> Yes this is possible for this you can take a look @ wifigrabber HID Script
Steps are as follows.
> 
> Find the Drive letter of your USB Drive via executing
((gwmi win32_volume -f 'label=''RPIW''').Name) in powershell
> 
> Where RPIW is the label of USB FLASH Drive you want to copy the files.
E:\ is the result in my case. Yours will be the exact drive letter of your Drive.
>
> then you can easily copy and past the files to that location.

I had such scenarios in mind when creating the UMS part of P4wnP1. Although I don't provide a ready-2-go payload, I provided this little tool with P4wnP1 A.L.O.A.:

https://github.com/mame82/P4wnP1_aloa/blob/master/dist/helper/genimg

The `genimg` script could be found at `/usr/local/P4wnP1/helper` and allows creating flashdrive or CDROM ISO images. Why is this worth mentioning ? Because the `-l` or `--label` command line flag allows to assign  a custom image name, which ends up as drive label when the volume is mounted by Windows. This equals to the `RPIW` label mentioned by @itsStar

mame82, 2019-01-28:
That's not exactly what I was thinking about in terms of automation ;-)

With `genimg` you're able to write a bash script, which:

- declares a variable with a custom drive label
- uses genimage to create the extracted image
- mount the image as flashdrive with P4wnP1_cli
- type out a PS script referring the label var to find the flashdrive and exfil data (again with P4wnP1_cli)
- unmount the image

Just an example

caretaker0815, 2019-01-28:
> [...] Find the Drive letter of your USB Drive via executing
> ((gwmi win32_volume -f 'label=''RPIW''').Name) in powershell [...]

This works great! :) 

>  [...] I provided this little tool with P4wnP1 A.L.O.A.:
> https://github.com/mame82/P4wnP1_aloa/blob/master/dist/helper/genimg [...]

I've created a 1GB .bin file with:
`dd if=/dev/zero of=test.bin bs=1000 count=1000000`

Now I have enough space to store the data and I know the drive letter.
Great! Many thanks for the help! :)

closed 2019-01-28 by caretaker0815

hommmi, 2025-04-18:
I've created a 1GB .bin file with:
dd if=/dev/zero of=test.bin bs=1000 count=1000000

and copied some files from PC to the sd card, how do I access these files using pi zero directly (ssh) what is the location of files that exist on the sd card (i cannot find it in home/pi or other folders tried searching using this command but did not find)
sudo find / -name "test.txt" 2>/dev/null