RoganDawes/P4wnP1_aloa #75: Payload injection

blankx, 2019-02-02:
Ive been banging my head against the wall for the last couple hours, curious if it's possible to inject msfvenom payloads if so, how

mame82, 2019-02-09:
Inject which payload format into what?

PHNTM-KEZ, 2019-05-24:
You might be able to do that. All you would need to do is generate the payload and store it on a partition created by ALOA (Mass Storage Emulation). Issue a move or copy command using HID script and execute accordingly, though most signatures are easily picked up from msfvenom, so obfuscation would be the real task.

21weberer, 2019-09-28:
I know this is kinda old, but cant you have a HIDScript paste the contents of a binary payload directly into CMD or Powershell and it will run without any need for obfuscation?