RoganDawes/P4wnP1_aloa #76: Victim cannot connect to internet through P4wnP1

mhjortsberg, 2019-02-05:
The victim Windows machine has the P4wnP1 connected to it, the Pi in turn connects to the attacking machine's WiFi hotspot to get internet access.

Problem is, the Pi gets internet connection, but the victim gets no internet connection even though it can ping the pi.

Tried to fix it by routing, bridging and iptables but nothing has worked.

mame82, 2019-02-05:
Not sure what you mean with victim. Thats not an issue with P4wnP1, nor is this an admin forum.

Let me try to help anyways, with an example of a possible configuration:

1) Assure your Windows host runs a DHCP client on the adigned USB network interface (default)
2) configure the DHCP server on P4wnP1 interface to hand out P4wnP1's USB IP as gateway and nameserver
3) Use iptables to set up masquerading on the upstream interface of P4wnP1 (wlan0 in your case)
4) Enable kernel ipv4 routing on P4wnP1, to make sure packets from USB end up in postrouting chain for masquerading

mhjortsberg, 2019-02-07:
Thanks for the advice. It turns out that i did not add the P4wnP1's IP address in the fields for DHCP and DNS server in the Pi's web interface. It was a bit confusing to understand that, but once i googled "RFC 2132" and found the table with codes from IANA, i got it working.
All the other steps you listed also has to be done.

closed 2019-02-07 by mhjortsberg

mame82, 2019-02-07:
I see.

Just for info:

For the default configuration I added an empty DHCP option 3 and 6, to explictly disable router and nameserver for DHCP leases. This is because dnsmasq would always hand out those two options for client leases, with the IP address of the DHCP server chosen as value for router/nameserver. This is not intended in every case, as one not always wants to hand out gateway/nameserver.

Reference

https://superuser.com/questions/306121/i-dont-want-my-dhcp-to-be-a-default-gateway

whackyhack, 2019-05-24:
Thanks for the ref.