RoganDawes/P4wnP1_aloa #88: Injecting mouse button clicks

monkeybus, 2019-03-07:
Hello there.

How can I inject mouse button clicks?

Moving the mouse is no problem, I cannot find out how to inject clicks.  Could you point me in the right direction?  Thanks a lot.

Great piece of software, by the way.

mame82, 2019-03-08:
Sorry for lack of documentation on this.
Methods are `click`, `doubleClick` and `button`.

They all accept arguments called `BT1`, `BT2` and `BT3` or combinations with logical or, like `BT1 | BT3`.

In contrast to the click methods, the `button` method keyps buttons pressed (for mouse dragging). A release could be done with `button(BTNONE)`

monkeybus, 2019-03-08:
Fantastic.  Works as advertised. Thanks a lot. I'll edit the readme tomorrow.

I am currently struggling with **absolute mouse positioning** _within Bash_.

I see the ms_snake HID script, and it makes sense, but could you let me know the Bash syntax?

I mean, given 1920x1080, how could I move the mouse to the bottom left corner of the screen?

I was thinking of P4wnP1_cli hid run -c 'move to(x,y);'  Of course, this doesn't work as I haven't declared the screen dimensions.  Am I in the right direction?  If you have time, could you let me know?

_Do you want me to put a wiki together?_  A lot of documentation needs to be collated.  Let me know, I'll get on it.

Thank you again.

Swiftb0y, 2019-03-10:
> Do you want me to put a wiki together? A lot of documentation needs to be collated. Let me know, I'll get on it.

I wrote the [legacy Wiki](p4wnp1.rtfd.io). Unfortunately, I am unable to rewrite because I have more urgent things to do. I would be grateful if you would continue to work on it. I've started an [aloa branch](https://github.com/mame82/P4wnP1-Wiki/tree/aloa) on the [git repository](https://github.com/mame82/P4wnP1-Wiki/tree/aloa) some time ago in an effort to revive it.

monkeybus, 2019-03-11:
OK, I'll take care of it.  I'll start on it this week.  

I'll probably have more questions.  If you have time, please do a quick answer and I'll keep adding to the wiki.

Thanks a lot for this.  I'll talk to you soon.

Swiftb0y, 2019-03-11:
Thank you for reviving the wiki. If you need a quick response, my [Twitter DMs](https://twitter.com/Swiftb0y_DE) are open.