RoganDawes/P4wnP1_aloa #89: External Power Supply

monkeybus, 2019-03-08:


felixperez, 2019-03-16:
Don't care. you can plug it first (or later) on a powerbank or in other USB of your computer.

Keep rocking!

mame82, 2019-03-16:
I had the same question when designing features whic are aimed at battery power buffer, while attaching to USB hosts from time to time.
Haven't found a satisfying answer myself in the specs. There was an old forum discussion on this topic, somewhere, but no reliable answer.

I'm using the a bunch of Pi0 with battery Power, while connected to host for two years without issues (no damage on hosts or devices). Yet, I won't give guarentees without having a technical proof.

What I can say for sure, is that the DWC2 kernel modul in use reports that power supply to USB port of Pi gets disabled, as soon as you plug it in to a host port with power supply (mode sense).

Maybe the answer resides in the specs of the DesignWare USB controller, still I wouldn't count on it, as it depends on hardware implementation of the Pi in the end.

Sum up: Sorry, I can't give a reliable answer, but it worked on all Pi0s I got my hands on during the last years, without any electrical damage.

Better comments on this issue are appreciated!!!

GaryH4, 2019-06-09:
I just bought a UPS HAT for pi0 called "UPS-Lite", which uses pins instead of USB ports.

And it may can also makes our p4wnPi more portable.

link: [https://www.aliexpress.com/item/New-UPS-Lite-UPS-Power-Board-Integrated-Serial-Port-Power-Detection-For-Raspberry-Pi-Zero-Zero/32954180664.html](url)