RoganDawes/P4wnP1_aloa #92: add keyboard layout

lapatte, 2019-03-14:
I try to add the Swiss keyboard because the .json file is not present in the image but it does not work so I import it into the keymap folder. When I go to look at the files present in the folder in question in ssh via putty the file already present inside its green while my ch.json file is written in white. is there any particular manipulation to do?

trifasico, 2020-01-01:
I have the exact same problem, any help?

carolinalopes98, 2020-01-01:
@lapatte hey, did you managed to solve this in the meantime? im getting the exact the same problem and i've already tried all the stuff i could think of and read in here but absolutely nothing seems to work

AdamWHY2K, 2021-05-23:
Green files are executable as you can see if you run `ls -l` I am not sure if files need to be executable to work as keymap files. To make your imported files executable too simply run `chmod +x ch.json`