RoganDawes/P4wnP1_aloa #93: HID blocked on startup

SeimuPVE, 2019-03-16:
On the web interface, the jobs of HIDs scripts are blocked on the state RUNNING and nothing appear on the target.

[image]

With the command line, it's blocked on "Start happening to 'HIDScript' in folder 'TMP'" : 
```
root@kali:~# P4wnP1_cli hid run -c 'type("Hello world")'
TempFile created: /tmp/HIDscript127740412
Start appending to 'HIDscript127740412' in folder 'TMP'
```

and nothing happened after that.

The HID keyboard and mouse are activated in USB Gadget Settings, many payload has been tried and there's not any reaction on the target.

Swiftb0y, 2019-05-20:
Not sure if this is the cause but are you sure you are using the correct usb-port?

HighSchoolDude, 2019-05-20:
I am having the same issue. I think it is a driver issue. under device manage I'm not seeing another device. I am made sure it plugged into the data port.

mame82, 2019-05-20:
The script seems to behave like intended.

The `waitLEDRepeat(NUM)` function in line 4 blocks execution till the NUM_LOCK LED is toggled frequently.

For a target OS with LED state shared between multiple keyboards (e.g. Windows) you could hit the NUM LOCK key of the real keyboard frequently (to change the LED). P4wnP1's virtual keyboard recognizes this LED change and continues execution.

So in simple words ... the hosts master keyboard could control the HIDScript.

mame82, 2019-05-20:
The `ms_paint` HIDScript is a nice example of the possibilities of this feature.

The script injects keystrokes to start paint and then takes over mouse control (but is blind, as it doesn't get feedback on mouse position). Now the LEDs of the master keyboard could be used to influence mouse behavior. 

Toggeling SCROLL LOCK pushes the mouse button (mouse cursor draws a line while moving). Toggeling NUM LOCK frequently turns the mouse movement direction left, hitting CAPS LOCK turns the direction right.

All in all, the script has complex behavior and an internal state, which is influenced from LEDs of the USB host, while the only connection between them is keyboard/mice

The script only starts execution, if NUM LOCK is pressed frequently in the beginning.

Here is a video demo of an early version:

https://youtu.be/zJ97CZJXtnA

PHNTM-KEZ, 2019-05-26:
I have the exact same issue. Nothing seems to be working, that is the HID interfacing/Keystroke injection isn't working. No driver installation either. Tried on both Ubuntu and Windows.