RoganDawes/P4wnP1_aloa #94: Event log not displaying anything

thaihugo, 2019-03-19:
Hi,

Thanks for your great work. I have the following bug: no event get logged in the event tab of the web client.

If I run P4wnP1_cli evt, I get some event printed. But I have nothing i the event log on the web client.

I installed aloa_alpha2, did apt upgrade, changed /etc/hostname and have a vanilla setup otherwise.

Can you point me to how the event log in the web browser gets it's message so I can try and debug it?

Thanks

thaihugo, 2019-03-19:
Of course, I have triggers setup to log event like join existing wifi or ssh user log, which are giving nothing

mame82, 2019-03-19:
Eventlog displays P4wnP1 internal events (f.e. TriggerAction with action Log).

thaihugo, 2019-03-19:
I try to trigger with ssh or join existing wifi. None of this event register anything. I'm sshing as root, and I suppose that maybe "join a wifi network" event activate before the logging is ready on the web client?
[image]

thaihugo, 2019-03-21:
Ok, I have a better grasp of what's happening. Basically, the event log will intercept an event if the web interface is currently connected with a client. It means USB event and probably gpio event can be intercepted if you are on the web client at the same time.

However, SSH event are not detected, and join a wifi network or create a wifi network might be created, but it would be useless as the web interface wouldn't be connected at the time. So the event might be displayed, but at a moment where you cannot see the interface, then when you can reconnect to it it would be cleared.

I also don't see how the "service started could be registered" as you cannot get to the client at the time.

Is there something I'm missing, and can you get this events logged and retrieved in any situation ?

thaihugo, 2019-03-21:
Note that when I have a Bluetooth connection on the device with a "P4wnP1_cli evt" command, it displays all event correctly, including ssh, although it seems to thing I'm connecting from the BTeth IP adress when the connection is coming from wifi. 
The web interface does not register any events. I'm on an Ipad if it changes anything.