RoganDawes/P4wnP1_aloa #97: gb Layout is currently broken

HazzaOnYT, 2019-03-26:
Hi, 

Hopefully this can be fixed, as I can't write scripts properly as it keeps using US layout. The problem with gb layout is it puts a weird character for e and a looks like a french letter.  The letter n is also missing. 

[image]

I have a separate question if for example you need to do a command like:
$example = "Joe Blogs"
How do I put this into HID, as the hid would be:
type("$example = "Joe Blogs"");  < However the issue is the " is counted anyway to stop this ?

Much appreciate all the help if anyone can!

HazzaOnYT, 2019-03-26:
I've tried to manually edit the gb.json , removing and adding a new E and N yet still doesn't make a difference.

mame82, 2019-03-27:
Not yet deployed to release image, GB layout has issues with redundant mappings ... see here

https://github.com/mame82/P4wnP1_aloa/commit/377f2e671624de79e6b53b0cf0719318dc1a714d

mame82, 2020-02-05:
fixed in current dev branch

tompusey, 2021-10-31:
How do I type a |? this may be an obvious thing but I can't for the life of me get it to output the pipe character in us or gb layouts?

tompusey, 2021-10-31:
> How do I type a |? this may be an obvious thing but I can't for the life of me get it to output the pipe character in us or gb layouts?

and as soon as I post it I found the answer, for me it works if you use "press('RIGHT_ALT ~')"