RoganDawes/P4wnP1_aloa #100: Trouble reverse engineering original P4wnP1 to ALOA

Mahendd, 2019-03-31:
The original P4wnP1 had some AMAZING demo payloads(mainly HID_Backdoor and the windows lockpicker) but the current ALOA has much better scripting, therefore, ideally, the user should be able to get the best of both worlds, and while that could, maybe, be accomplished by installing both versions(not that I have any idea of what would happen), a probably much better idea is to reverse engineer the original payloads, and that's what I decided to do.
    However, it soon proved to be quite harder than my inexperienced me thought it was due to the connection between multiple files and dlls(some of which even apparently had unrecognizable characters, possibly German).
    So here it comes to the actual issue: Has anyone ever "translated" the payloads? Is anyone trying to? Am I missing something on the readme or the files?
    Or if anyone is more experienced, be it on reverse engineering or even the P4wnP1 itself, could you give me tips towards finding what I seek? Thank you.

mame82, 2019-03-31:
P4wnP1 A.L.O.A. could accomplish every task, which was accomplished by P4wnP1.

The only exception is the HIDBackdoor, which was a seperate piece of software. The reason I haven't ported it to A.L.O.A. (yet), is that the backdoor server heavily relied on the old HID scripting language (python parser for DuckyScript vs new JavaScript based interpreter).

Anyways, there's not much to reverse, as the whole backdoor was open sourced

mame82, 2019-03-31:
Client

https://github.com/mame82/P4wnP1_HID_backdoor_client

mame82, 2019-03-31:
Server

https://github.com/mame82/P4wnP1/tree/master/hidtools/backdoor