RoganDawes/P4wnP1_aloa #103: Error while deploying new USB gadget settings rpc error

satnavboy100, 2019-05-15:
So when i try to setup the mass storage on the pi it gives me this error and im not sure what to do, i have tried this on 2 separate raspberry pi zero w's and they both give me the error. 
 
"Error while deploying new USB gadget settings rpc error: code = Unknown desc = Deploying new gadget settings failed, reverted to old ones: Gadget Settings consume 8 out of 7 available USB Endpoints%0A

if any one has any ideas please let me know

mame82, 2019-05-15:
You should disable one of the other gadgets, as the USB hardware layer is
limited by available endpoints and each gadget consumes several of them

satnavboy100, 2019-05-15:
Ah ok thank you, I installed the os and put the usb cable into the correct usb port but it dosent seem to be working when ever I do a HIDScript I have tested it in both Linux and windows but I'm windows it shoes that it dosent recognise the device and it dosent show up as an RNDIS device either so I'm not sure what's going on, once again I have tested this on 2 raspberry pi zero w's and the error is still there, I tried it in Linux to see if that would change anything but it dosent, I don't know if I am missing anything but I did everything step by step

closed 2019-05-15 by satnavboy100