RoganDawes/P4wnP1_aloa #104: Can't connect to wifi

greatod, 2019-05-17:
Hi,
First let me say thank you for this piece of software.
I'm trying to get the P4wnP1 to connect to the Wifi. I tried failover to AP, but it allways reverts to AP. Got the usbeth working in order to connect via SSH, and tried this: 
P4wnP1_cli wifi set sta -n -s DivineNet -k somepsk
and i got this:
Deploying WiFi inteface settings:
        name:"default" regulatory:"US" working_mode:STA ap_BSS:<> client_BSS_list:<SSID:"DivineNet" PSK:"somepsk" >
Non of the given SSIDs found during scan
Could you help me diagnose this problem?
Thank you

greatod, 2019-05-19:
I figured it out.
My router was broadcasting on channel 13. The Raspberry pi zero W that i used was unable to scan channel 13. Changed the channel of my router to 11, and now everything is ok.

mame82, 2019-05-19:
Be sure to use the proper regulatory domain. Channel 12,13,14 aren't part of 'us' regulatory. If you  choose a proper regulatory domain, it should work with channel 13, too.

greatod, 2019-05-19:
Ok. Didn't know that...Thank you.

mame82, 2019-05-19:
Nobody knows everything. To be honest, I added the option to set regulatory
domain, because I ran into the same issue myself