RoganDawes/P4wnP1_aloa #105: No wifi access point

DualByte, 2019-05-18:
I dont anything why it doesnt work

lukespademan, 2019-05-19:
I am also having issues. I have flashed 0.1.0-alpha2 using win32 disk imager and after waiter for 5mins none of my devices have detected any beacon frames (network does not appear in list). I am using:
* Raspberry Pi Zero w V1.1
* sandisk Ultra 16GB class 10 micro SD card

@DualByte is this what you are also experiencing? Can you please give any other infomation about your issues to improve the chances of this being resoled.

DualByte, 2019-05-19:
> I am also having issues. I have flashed 0.1.0-alpha2 using win32 disk imager and after waiter for 5mins none of my devices have detected any beacon frames (network does not appear in list). I am [...]

I am having the same probmen and Im actually using even the same SD card as you. I dont know anything why it wouldn't work.

HighSchoolDude, 2019-05-20:
@DualByte  I doubt this would be causing the problem but have you tried using a different ISO software like Etcher, or Rufus?

DualByte, 2019-05-20:
> @DualByte  I doubt this would be causing the problem but have you tried using a different ISO software like Etcher, or Rufus?

No I have not actually tried that. I will try it shortly

Swiftb0y, 2019-05-20:
I didn't have a good experience with Etcher. I recommend Rufus on Windows or just plain old dd on macOS and Linux.

DualByte, 2019-05-20:
> @DualByte I doubt this would be causing the problem but have you tried using a different ISO software like Etcher, or Rufus?

Rufus didnt work, either win32diskimager or etcher was working.

lukespademan, 2019-05-20:
I used etcher and it now works :)

closed 2019-05-20 by mame82

DualByte, 2019-05-20:
I didnt get it to work

lukespademan, 2019-05-20:
Have you tried using etcher? Do you have a different microSD you can try?

DualByte, 2019-05-20:
> Have you tried using etcher? Do you have a different microSD you can try?

I have tried and also have tried using other systems like raspbian, ubuntu mate but it have not show the wlab0 interface and it has been working before but not anymore.

lukespademan, 2019-05-20:
Is it a pi zero or pi zero w?

DualByte, 2019-05-20:
> Is it a pi zero or pi zero w?

It is pi zero w

lukespademan, 2019-05-20:
This sounds like a hardware issue and is outside the scope of this project. I can advise you look at the raspberry pi forums for advice :)

DualByte, 2019-05-20:
> This sounds like a hardware issue and is outside the scope of this project. I can advise you look at the raspberry pi forums for advice :)

Okay. Thanks for the help.