RoganDawes/P4wnP1_aloa #108: Fetch storage files from browser

danthegoodman1, 2019-05-19:
A really nice feature would be to view/download the files in the mass storage from the admin panel. Or even having new ones be uploaded to a bucket/server when added.

Maybe the first one is possible and I am just being absolutely blind.

TheDevMinerTV, 2019-05-19:
You should be able to copy the storage from the connected computer's drive via HIDScript to your Pi's storage.
An example on how to copy files to the storage created by P4wnP1 ALOA by Seytonic [**here**](https://www.youtube.com/watch?v=I_BjCdJlCo4).

danthegoodman1, 2019-05-19:
@TheDevMinerTV yes, but that’s not what I mean. I am referring to once you get the files onto the storage using HID that you can view/download them from the admin webpage so you don’t have to remove the pi and plug it into your device to fetch the files.

TheDevMinerTV, 2019-05-19:
Okay then, never mind

mame82, 2019-05-19:
For now I won't implement features into the web client, which are available
via ssh/scp already.

Suggested solution:

Once the flash flashdrive is uncounted from the USB host, loop mount the
backing image to P4wnP1's file system. Then access the files via SSH/SCP.
(Could be automated with bash scripts and started using TriggerActions if
you like).

Be sure not to mount the image to Pi while the flashdrive is in use: Linux
kernel docs (USB UMS gadget) make clear that this leads to inconsistent
state (and it does, I have tested).

Cheers

Mai 2019, 18:57:

danthegoodman1, 2019-05-19:
@mame82 Thanks for the suggestion! Maybe I’ll take a look into adding them in to the web client when I’m done with finals.

mame82, 2019-05-19:
The web client is basically a client for a grpc server written in gopherjs (a proprietary version of grpc-web to be more specific). Client code runs directly in the browser as SPA and calls RPC methods of the backend service. The latter doesn't provide file IO remote methods (or only limited ones for transfer of HIDScript content). This is to avoid compromising P4wnP1's file system, as the client wasn't built with security in mind. So as stated SSH/SCP does a  better job for this. If you have the feeling you need another form of file access, adding another web-based file browser to the startup configuration is perfectly fine (f.e. webdav), but I want do so. Don't get me wrong, but I won't implement features into RPC server and RPC clients (web/cli), which could be provided by common external tools