RoganDawes/P4wnP1_aloa #109: Read out the connected device's operating system

TheDevMinerTV, 2019-05-19:
Is it possible to get the operating system of the connected computer from HIDScript to type something different for Windows than Linux, for example?

Swiftb0y, 2019-05-20:
@jcstill might be able to help you in that regard.
Also take a look at [this issue](https://github.com/mame82/P4wnP1/issues/206).

mame82, 2019-05-25:
Topic is alive, currently discussed on Twitter 

https://twitter.com/JoelSernaMoreno/status/1132192054824587264?s=19

jcstill, 2019-05-26:
@Swiftb0y thanks for tagging me in this. I completely forgot about this project.
I think I will start up work again now that I'm done with classes for the summer.

mame82, 2020-06-15:
Helpful input from @abarisani

https://gist.github.com/abarisani/4595a7c535435038e0571237893c81c4