RoganDawes/P4wnP1_aloa #110: [USB MASS STORAGE] USB mage not mounting

HighSchoolDude, 2019-05-20:
When ever I make a new image using genimg it work and this is the output 

`Generating 1000MB FAT32 image at /usr/local/P4wnP1/ums/flashdrive/1gigabyte.bin
1000+0 records in
1000+0 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 61.1812 s, 17.1 MB/s
mkfs.fat 4.1 (2017-01-24)`

But when I choose the image in the web UI the drive doesn't mount on the PC, I also checked the /mnt folder to see if it was my computer but there is nothing there.

HighSchoolDude, 2019-05-20:
I ran the P4wnP1_cli script and I got this error message.

`2018/12/07 01:51:51 Error deploying current USB Gadget Settings: rpc error: code = Unknown desc = Deploying new gadget settings failed, reverted to old ones: if the composite gadget isn't disabled, as least one function has to be enabled
Error deploying Gadget Settings: rpc error: code = Unknown desc = Deploying new gadget settings failed, reverted to old ones: if the composite gadget isn't disabled, as least one function has to be enabled
Reverted to:
enabled:true vid:"0x1d6b" pid:"0x1347" manufacturer:"MaMe82" product:"P4wnP1 by MaMe82" serial:"deadbeef1337" use_CDC_ECM:true use_RNDIS:true use_HID_KEYBOARD:true use_HID_MOUSE:true rndis_settings:<host_addr:"42:63:65:12:34:56" dev_addr:"42:63:65:56:34:12" > cdc_ecm_settings:<host_addr:"42:63:66:12:34:56" dev_addr:"42:63:66:56:34:12" > ums_settings:<> dev_path_hid_keyboard:"/dev/hidg0" dev_path_hid_mouse:"/dev/hidg1" `

mame82, 2019-05-20:
I'm not exactly sure which parameters you provided to the `P4wnP1_cli` client binary, but it seems you enabled no single USB service. You can't bring up an USB composite device with no single interface enabled.

SirJan18, 2019-05-20:
@mame82  I was using the Hello World command `P4wnP1_cli hid run -c 'type("Hello world")'`. 
How would I enable a USB service to, for example, make this script work?

SirJan18, 2019-05-20:
I was attempting to mount an image named 500mb.bin I made with genimg, I will put search for the command and send it once I find it. Whenever I try using the web UI it says successfully mounted but there I no disk on the target PC. I checked with Diskpart and the Create and Partition Disk tool on Windows and there is nothing.

I have tried to troubleshoot and find the cause but to no avail, is there any file output you would need to help with the troubleshooting process?

SirJan18, 2019-05-21:
Turns out the cable was not providing a data connection, I switches the cable and it works now.

closed 2019-05-21 by SirJan18