RoganDawes/P4wnP1_aloa #111: Can this work on the pi 3?

jakel181, 2019-05-21:
Title says it all. I couldn't think of a better place to ask this question than here.

Soltys195, 2019-05-21:
It dont see why it would not work as pi 3 has better hardware than pi zero.

jakel181, 2019-05-21:
> It dont see why it would not work as pi 3 has better hardware than pi zero.

That's good. :) I can't wait to try it out, and don't want to wait for shipping of a Pi zero w  :P

mame82, 2019-05-21:
No it doesn't work on Pi3

whackyhack, 2019-06-14:
P4wnP1 (including ALOA) is heavily USB centric in its current design.  Classic Pi's Micro USB socket is wired for power only even though their hardware are capable of OTG.  As such, functionality will be very limited, if any.

As of now, P4wnP1 ALOA does not activate Pi3's WiFi, nor its Bluetooth.