RoganDawes/P4wnP1_aloa #116: CRC Checksum for img.xz

Benderz, 2019-05-24:
Windows image writers and unpackers give me an error about the corrupted "archive kali-linux-v0.1.0-alpha2-rpi0w-nexmon-p4wnp1-aloa.img.xz". Downloaded it many times - same error.

whackyhack, 2019-05-26:
I agree that some form of verification should be posted.  Meanwhile, you may be able to download the p4wnpi image under https://www.offensive-security.com/kali-linux-arm-images/#1493408272250-e17e9049-9ce8 where you can find sha256 digest. (The posted file size is different from the one on github.)

Jaimes28, 2019-06-28:
The same thing happens to me on windows when I go to flash it to the micro sd it doesn't show up and I do not have the option to flash it onto the micro sd