RoganDawes/P4wnP1_aloa #118: Rpc Error (Mouse & Keyboard) Disabled & Storage Emulation Failure

PHNTM-KEZ, 2019-05-24:
Cannot run any HID scripts, from the Browser GUI or the command line. (Tried running a "type" command using the P4wnP1_cli via SSH, as seen in README.md. Gave me an error). Don't receive any notifications about driver installation upon plugging it in (Windows 10). "Test.bin" does not create any storage options, tried both in Ubuntu and Windows. All default values for USB ID changed from defaults.

Command: 
sudo P4wnP1_cli hid run -c 'type("Please just  fucking work")'

Terminal Output: 
TempFile created: /tmp/HIDscript759608868
Start appending to 'HIDscript759608868' in folder 'TMP'
2019/05/17 20:30:25 rpc error: code = Unknown desc = HIDScript not available (mouse and keyboard disabled)

Tried enabling via browser GUI. Says successful USB gadget changes, GUI registered the toggled settings. Command line gives me the same error afterward.

iAMxGiGO, 2019-07-30:
This is also happening to me and for some reason as soon as i deploy my template it resets back to the startup template. I tried being a smartass and changed the startup template to my needs and it still resets to the default template. I also tried making a custom master template but that also resets. As of now i can't really use p4wnp1 the way i want and i have no clue what could be causing this. The weird part is that when i deploy i literally see the settings change to my template and then back to the default.

JimKoML, 2022-11-19:
Hi, did you resolve this issue? I am also facing the same problem.

xyon1282, 2023-02-25:
Hi, did you mange to resolve this? I'm having the same issue here

PHNTM-KEZ, 2023-02-25:
Hey, unfortunately no, I was unable to solve the issue. I landed up
purchasing a standalone keystroke injection device, similar to a
rubber-ducky.

jakobkarosseit, 2023-04-22:
@PHNTM-KEZ which device did you buy?