RoganDawes/P4wnP1_aloa #119: After Flashing No Detection of Wifi Network nor anything

brijq, 2019-05-26:
Thank you for the great software. However after flashing from https://www.offensive-security.com/kali-linux-arm-images/ , pluging in to windows pc or Mac OS none can detect any sense of the network.

I tried using ssh with username root also timeout. Please assist

brijq, 2019-05-26:
Please note that I downloaded this one under raspberry pi foundation tap and under Pi Aloa image

whackyhack, 2019-05-26:
Maybe you can download directly from "Releases" on github.  The file size listed on Kali site looks different.

brijq, 2019-05-26:
I tried the 'releases' on GitHub same no detection nothing.

whackyhack, 2019-05-26:
I flashed my PiZero. (My first Kali.  Yay!)  Using an rtl1892 dongle, I am able to get onto the p4wnp1 WiFi network.  But I, too, do not get OTG Ethernet. (I am using MacOS.)  Previously, I was able to get OTG Ethernet on Raspbian.

I can see that u_ether is loaded.

```
Module                  Size  Used by
ccm                    10688  6
bridge                119099  0
stp                     2126  1 bridge
llc                     5343  2 bridge,stp
usb_f_hid              12455  7
usb_f_ecm               7454  2
usb_f_rndis            17958  2
u_ether                12869  2 usb_f_ecm,usb_f_rndis
bluetooth             370503  2
ecdh_generic           18156  1 bluetooth
libcomposite           48920  15 usb_f_ecm,usb_f_rndis,usb_f_hid
rtl8xxxu               99684  0
arc4                    2332  2
rtl8192cu              77906  0
rtl_usb                11748  1 rtl8192cu
rtl8192c_common        53645  1 rtl8192cu
rtlwifi                87834  3 rtl_usb,rtl8192c_common,rtl8192cu
mac80211              727794  4 rtl_usb,rtlwifi,rtl8192cu,rtl8xxxu
sg                     21423  0
cfg80211              542587  2 mac80211,rtlwifi
rfkill                 21451  4 bluetooth,cfg80211
uio_pdrv_genirq         3686  0
fixed                   3001  0
uio                    10016  1 uio_pdrv_genirq
dwc2                  131147  0
udc_core               38881  6 usb_f_ecm,dwc2,u_ether,usb_f_rndis,libcomposite,usb_f_hid
ip_tables              12532  0
x_tables               22404  1 ip_tables
ipv6                  404418  31 bridge
```

brijq, 2019-05-26:
I just want to clarify that the proper way is that after you flash the image straight away you plug towards your pc it will auto spawn the p4wnp1 WiFi network? And does it work on macOS or it must be a windows machine?

whackyhack, 2019-05-26:
> plug towards your pc it will auto spawn the p4wnp1 WiFi

O, I couldn't power Pi from my Mac because mine was a lowly Zero, not Zero W, so I needed that USB OTG port to hook my WiFi dongle.  I powered it from the power-only USB Micro connector (PWR IN).  WiFi shouldn't care whether it's Windows or Mac. (USB-Ethernet, on the other hand, could have different results because Windows only supports RDNIS.  Luckily, P4wnP1 supports it.)

However, I did initially try to power from OTG without dongle, expecting USB-Ethernet to get me in.  When I couldn't get an Ethernet interface, I thought Kali failed to complete boot due to power limit on my Mac port, or due to excessive voltage drop on my cable (which was frequently an issue with CHIP).  But that theory does not seem to hold because I have had no problem getting Raspbian to power Pi Zero via OTG while offering USB-Ethernet. (All Pies seem to draw less current than CHIP at boot time.)

When I tested USB-Ethernet on Raspbian, I was using g_ether as common guides suggested.  I don't know its exact relationship with u_ether, although I also tried to load g_ether from the boot loader.   g_ether loads, usb_f_rdnis and u_ether also find it.  But I still cannot see Ethernet from Mac.

```
Module                  Size  Used by
ccm                    10688  6
usb_f_hid              12455  0
bluetooth             370503  2
ecdh_generic           18156  1 bluetooth
rtl8xxxu               99684  0
arc4                    2332  2
rtl8192cu              77906  0
rtl_usb                11748  1 rtl8192cu
rtl8192c_common        53645  1 rtl8192cu
rtlwifi                87834  3 rtl_usb,rtl8192c_common,rtl8192cu
mac80211              727794  4 rtl_usb,rtlwifi,rtl8192cu,rtl8xxxu
cfg80211              542587  2 mac80211,rtlwifi
rfkill                 21451  4 bluetooth,cfg80211
fixed                   3001  0
uio_pdrv_genirq         3686  0
uio                    10016  1 uio_pdrv_genirq
usb_f_ecm               7454  1
g_ether                 4973  0
usb_f_rndis            17958  2 g_ether
u_ether                12869  3 usb_f_ecm,g_ether,usb_f_rndis
libcomposite           48920  12 usb_f_ecm,g_ether,usb_f_rndis,usb_f_hid
dwc2                  131147  0
udc_core               38881  6 usb_f_ecm,dwc2,u_ether,usb_f_rndis,libcomposite,usb_f_hid
ip_tables              12532  0
x_tables               22404  1 ip_tables
ipv6                  404418  30
```

(When I do this, Mac complains that the USB power load is too high so I couldn't even continue without  external power.)

mame82, 2019-05-26:
Thx for valuable input.

1) P4wnP1 A.L.O.A. only supports Pi0W as the backing service expects communication from the modified `bcmfmac` driver (which isn't loaded if the Broadcom WiFi chip isn't present)

2) Although I can't test against a Mac frequently, legacy P4wnP1 was tested to some extend by its users. When it comes to port based Power supply, A.L.O.A. reports the exact same MaxPower value via USB descriptor as its predecessor (250mA max currency).

Reference: https://github.com/mame82/P4wnP1_aloa/blob/master/service/SubSysUSB.go#L46

If external devices are stacked onto the Pi, it is likely that more than the specified currency is taken from the port. If the Mac complains on this, Apple did the job right.

brijq, 2019-05-27:
My MacBook Pro is running on Broadcom BCM43xx 1.0 (7.77.61.2 AirPortDriverBrcmNIC-1305.8) Chip however it still does not detect the wifi?

Sorry I am new to P4wnP1 A.L.O.A. please help me out if I am not on the right track. :)

whackyhack, 2019-05-28:
The comment about chip was regarding Pi0W, not the target platform.

On USB-Ethernet, I find Mac more friendly.  For example, it supports CDC composite out of the box, which I believe comes from Linuxland.  I just downloaded the Raspbian P4wnP1, and it is able to talk to Mac. (My Windows box is not easily booted but I believe that it would work as well.)  The puzzle is that the drivers look the same.
```
Module                  Size  Used by
usb_f_hid              11655  2
usb_f_ecm               7416  2
usb_f_rndis            17119  2
u_ether                13511  2 usb_f_ecm,usb_f_rndis
snd_bcm2835            23131  0
snd_pcm                97889  1 snd_bcm2835
snd_timer              22706  1 snd_pcm
snd                    68784  3 snd_timer,snd_bcm2835,snd_pcm
dwc2                  122113  0
bcm2835_gpiomem         3791  0
uio_pdrv_genirq         3718  0
uio                    10166  1 uio_pdrv_genirq
fixed                   3029  0
libcomposite           49907  14 usb_f_ecm,usb_f_rndis,usb_f_hid
udc_core               40452  6 usb_f_ecm,dwc2,u_ether,usb_f_rndis,libcomposite,usb_f_hid
ip_tables              12512  0
x_tables               20921  1 ip_tables
ipv6                  384901  12
```

whackyhack, 2019-06-02:
Just to close out my own notes: After plugging the same SD card into my new Pi Zero W, my Mac recognizes the USB CDC/ECM gadget, as well as the USB HID gadget; WiFi AP also works. (No BT device is detected.)  In other words, the image works out of the box on rpi0w.  I also tested the image from Kali site with the same result. (A third image I tested was the legacy P4wnP1.)

This said, if a USB serial console can be forced on externally (via cmdline.txt, for example) when all network interfaces fail, the user will be able to logon and do more diagnosis. (Currently, in order to enable USB-serial interface, one must first gain network access, disable HID interfaces, only then enable serial.  Even then, a TTY has to be started separately on this new serial interface.)

mame82, 2019-06-02:
Correct, tty has to be started manually for the serial interface. Reason: there are other use cases for CDC ACM, beside terminal CLI (f.e. raw data transfer).

Anyways, I wonder why HID gadgets should be disabled. This is not a requirement. Only limitation is, that enabled gadgets mustn't consume more than 7 USB endpoints all together (error message is shown for such a configuration)

whackyhack, 2019-06-03:
Regarding serial TTY, my thinking is more about friction in user workflow.  As serial is not a default at start, which use case to require an extra step after being enabled becomes another point of default.  To a new user, especially if Ethernet and WiFi appear unavailable, having an immediate console is extremely valuable. (Provided it is possible to enable serial at boot time.)  To a sophisticated user that needs raw data, one extra step through an established connection would be tolerable.

Of course choice of default can be argued with probability of use, about which I have no data.

(I filed the observation about conflict with other OTG gadgets in #131, then an issue with post-boot OTG plugin in #132.)

mame82, 2019-06-03:
Serial could be default (store USB config with serial as startup template, or modify the Startup MasterTemplate to use an ISB template with serial enabled).

A bashscript which adds try to the serial could be added to a TriggerAction, which fires "on service start". After adding the respective TriggerAction template to the Startup MasterTemplate, it will survive boot, too.

So in summary, the behavior you are aiming at could be achieved by editing the Startup MasterTemplate to your needs... exactly what it was integrated for

whackyhack, 2019-06-05:
I agree that **if** the user has gained connection into the system, all these can be done with relative ease.  Probably I should have said "life cycle" instead of "work flow" because I am thinking about @brijq's situation, i.e., before network logon becomes possible.  A serial console would have allowed access to some debugging info like dmesg, syslog, and command output.

Meanwhile, the "on service start" trigger could potentially be added by mounting the ext4 file system on a workstation.  It just takes more than a beginner's toolkit to do this type of thing. (Anyone knows how to edit a file in the FAT boot FS because it automatically mounts on every popular OS.)

mame82, 2019-07-24:
Note: statements on USB CDC ACM should be included into a FAQ