RoganDawes/P4wnP1_aloa #121: USB is not working

chazp246, 2019-05-26:
Sooo.
I downloaded aloa from kali.org.
It boots up shows wifi but win 10 show unknow device and i cant do anything.
I can go to webinterface and do what i want but usb emulation just dont work.
When i try to change settings of usb it show it was succesful but windows will still show uknown device.
Cable is good.
I have usb directly soldered on rpi 0 w but when i first soldered it i swap vcc and gnd.
Is it possible that i burned usb controller chip?

PHNTM-KEZ, 2019-05-26:
Same thing has been happening to me. The test.bin file doesn't create a storage option for me. 

Try reverse engineering to use method of elimination? Connect via mini hdmi, or ssh in somehow. Connect a USB stick to the data port and use "sudo fdisk -l". Mount to the stick and transfer/create a file. If everything goes well, the controller should be fine. If not, then something might be up.

Just guessing, far from a professional.

mame82, 2019-06-02:
You can't assure the controller is working fine, by connecting an external USB device.

The DWC2 USB controller internally consists of a hybrid solution with two parts:
1) USB device controller (used by P4wnP1's USB gadgets)
2) USB host controller (active when an external USB device is connected)

The DWC2 automatically decides which mode to use (mode sensing) and disables unneeded controller parts. Both models can't be active on the same time (at least at controller firmware level).

Testing with an external USB device brings the controller into USB host mode, which means the results don't apply to USB device mode (used by P4wnP1)

chazp246, 2019-06-03:
@mame82 thanks it can't load up keyboard or any USB flashdrive. I Will be buying new rpi0w

whackyhack, 2019-06-14:
> It boots up shows wifi but win 10 show unknow device and i cant do anything.
> I can go to webinterface and do what i want but usb emulation just dont work.
> When i try to change settings of usb it show it was succesful but windows will still show uknown device.

I see the same "unknown device" message in win10 but find it harmless. "Unknown" is the extra interface usb1 with CDC ECM protocol that Windows doesn't speak. (See #130.)  When you say "I can go to web interface," can you connect via 172.16.0.1:8000 from the victim machine?  If you can, that is a sure sign that USB gadget is working.

whackyhack, 2019-06-14:
> It boots up shows wifi but win 10 show unknow device and i cant do anything.
> I can go to webinterface and do what i want but usb emulation just dont work.
> When i try to change settings of usb it show it was succesful but windows will still show uknown device.

I see the same "unknown device" message in win10 but find it harmless. "Unknown" is the extra interface usb1 with CDC ECM protocol that Windows doesn't speak. (See #125.)  When you say "I can go to web interface," can you connect via 172.16.0.1:8000 from the victim machine?  If you can, that is a sure sign that USB gadget is working.

dopescope45, 2019-07-21:
i look in the devices menu and it says no driver installed what do i do

whackyhack, 2019-07-22:
@dopescope45 no need to look at the device showing "no driver installed" - that is just an interface that Win10 is unable to use.  Look for the Ethernet interface that wasn't there without p4wnp1_aloa.