RoganDawes/P4wnP1_aloa #122: RNDIS No connection from pi to clients

ducng99, 2019-05-27:
I have setup RNDIS and wifi in DHCP_SERVER mode, my target can connect to the pi but my pi cannot reach the target, can't even ping. Is this a limitation?
I'm using default settings: Pi: 172.16.0.1 - Client: 172.16.0.2

mame82, 2019-06-02:
Sounds like a firewalling issue. Assuming the host is Windows, try to classify the network as private instead if public (Windows won't send ICMP replies otherwise).

If the target could connect to Pi, you already have communication in both directions. Target likely runs a connection tracking enabled firewall

ducng99, 2019-06-02:
Yes it is firewall issue, I disabled firewall for public network and it's working properly now.
Thank you!

whackyhack, 2019-06-02:
This is amazing - or puzzling.  Is this a corporate policy to classify 172.16/16 as public network or something that a vanilla Windows will do?

ducng99, 2019-06-02:
For Windows (vanilla), any new networks will be treated as public. Once connected, it will ask users whether they allow sharing info and stuffs, this should switch the network to private but sometimes it won't, you must go to network's properties and change to private manually.

In this case, RNDIS only act as network interface without internet connection (default settings), windows will not allow you to change to private (in my case there aren't any options). What I did was to go to Control panel and disable firewall system-wide.

However different versions of windows may behave differently, I'm using Win 10 Pro 1809.

whackyhack, 2019-06-05:
Now that you explained it, I finally understood why a vendor asked us to disable host firewall on a Windows 201x Server when things didn't talk. (That was my first Windows server in our Linux shop.)

mame82, 2019-08-04:
Here's a video on legacy P4wnP1, which covers a totally different topic (Internet Connection Sharing). Amongst other things, the video shows how to change the Profile of a Windows network connection from "public" to "private", using PowerShell.

https://youtu.be/QEWaIoal5qU

I highly recommend NOT TO DISABLE the Windows Firewall for public networks