RoganDawes/P4wnP1_aloa #125: Driver not available Win10 with latest Updates

svenpanke, 2019-05-30:
When i plug in the P4wnP1 ALOA Stick it boots up, i have acess to the Control Panel over USB and Wifi too.
But it seems that Windows 10 does not find the driver...
I have testet it on two different PCs.
Has anyone else the same problem?
[image]

mame82, 2019-05-30:
Windows has no CDC ECM support. If you are targeting Win only, just disable the CDC ECM interface in USB settings. Windows uses RNDIS. 

Be sure to grant a different product ID or vendor ID when disabling/enabling USB functions, to avoid issues with cached configurations on Windows end

whackyhack, 2019-05-31:
I had been frustrated by this compatibility gap on another gadget platform.  This means that one cannot make an out-of-the-box gadget.  I wonder if it is possible to fingerprint the target in some other way and automatically unload/reconfigure. (Thanks for the product ID/vendor ID tip.  It is sneakier than cleansing the target.)

whackyhack, 2019-06-02:
> I wonder if it is possible to fingerprint the target in some other way and automatically unload/reconfigure

The following comment in the legacy P4wnP1 answers that question:
```
	# Note: Detection for RNDIS (usb0) is done first. In case it is active, link availability
	#	for ECM (usb1) is checked anyway (in case both interfaces got link). This is done
	#	to use ECM as prefered interface on MacOS and Linux if both, RNDIS and ECM, are supported.
```
In other words, yes, it is possible.  If usb1 is torn down when RNDIS is detected, Windows should not complain.

mame82, 2019-06-02:
ALOA follows a slightly different approach.

If RNDIS and CDC ECM are enabled, two USB ethernet interfaces exists on Pi0W end.

Legacy P4wnP1:

- configures each internal interface (dedicated IP, dedicated DHCP etc)
- tries to disable unneeded interface based on naive logic, to avoid conflicts (layer 2 connection polling)
- even one if the two interfaces is disabled, they are still visible to the USB host

A.L.O.A.:

- both internal Interfaces are bridged together to a new virtual interface `usbeth`
- a single configuration (DHCP, IP etc) is applied to the virtual interface
- the user should manually disable unneeded USB ethernet gadgets (no CDC ECM needed for Windows)
- if both are enabled,though, same IP settings apply to Windows, Linux, Mac (although the host doesn't install drivers for unsupported interfaces) ... this avoids distinguishing IP configurations during scripting for different USB host OSes

mame82, 2019-06-02:
For details on cached USB descriptors for Windows, see my comments in legacy P4wnP1 source:

https://github.com/mame82/P4wnP1/blob/master/boot/init_usb.sh#L150

Caution: I don't give support for manual registry manipulation, as it could easily mess up Windows USB stack (had hard times with this myself). So the easiest thing one could do, is to use a dedicated USB PID per gadget configuration.

whackyhack, 2019-06-09:
I managed to connect my broken Win10 laptop to TV and plugged in p4wnp1.aloa.  A transient message says "Installing new device P4wnP1", then "Your device is ready to use."  If I go to Control Center, there is indeed a P4wnP1 device with the note "No driver available".

On the other hand, I can verify that the RDNIS interface is fully functional with a p4wnp1 assigned address. (Same as @svenpanke observed.)  In other words, this "unknown device" is not an impediment. (It also disappears once p4wnp1 is detached.)