RoganDawes/P4wnP1_aloa #126: Custom USB images not working properly

Droggelbecher42, 2019-05-30:
I'm having trouble with the custom images for larger simulated USB mass storage devices.
To get this out of the way: the test.bin works fine, it shows up as "README" and I can use it. I succesfully created new images with various sizes (10 MB to 2GB) and different names (all lowercase, all caps etc) using the script provided. The images show up in the ALOA interface. At first they were not recognized by my host (Win 10) at all. I then discovered that the fake product and vendor IDs seem to interfer with that (I had them set to fake an iPhone 5S). With the default settings for those, the mass storage now shows up in Windows, but with the wrong name! No matter what I assigned during the image creation, it is always called "P4wnP1 ALOA". I tried setting the fake IDs to match a Kingston Datatraveller USB stick, but that had no effect (still wrong name).
At this point I'm kind of out of ideas how to fix this. I don't think it's my USB cable either, since it's from a smartphone so it should have data. Also, the test.bin image works just fine. Don't know if this is related or just normal behaviour: during bootup of the pi, the windows sounds indicate that a device connects and disconnects multiple times before the drive finally shows up.

Droggelbecher42, 2019-05-30:
might be related to this issue: https://github.com/mame82/P4wnP1_aloa/issues/125 ?

Droggelbecher42, 2019-05-30:
Another clue for what might be going on: when the USB storage device connects as "P4wnP1 ALOA", the HID script apperently gets messed up, too. Instead of GUI+R it seems to do CTRL+F as the little search feature in my browser pops up and the commands seem to be typed in there before it quickly disappears again. So I guess something with the whole device emulation is messed up when that happens. The HID script works just fine when I use the test.bin image.

mame82, 2019-06-02:
Have you tried to change USB PID when you use new USB settings (see comments on #125 for Windows USB descriptor cashing issues)

Droggelbecher42, 2019-06-04:
> 
> 
> Have you tried to change USB PID when you use new USB settings (see comments on #125 for Windows USB descriptor cashing issues)

I tried it now, with no success unfortunately. I also tried disabling CDC ECM, disabling both the HID options and changing around the MAC adresses for the RNDIS connection, all with no effect. The storage always shows up as "P4wnP1 ALOA". I also tried whether it makes a difference if I'm using a USB 2 or 3 connection, it does not.

Again, I noticed that the HID script gets broken, as it starts performing all sorts of actions on the browser window if it is open instead of opening a commandline.

Next thing I will try is deploying it on a different Windows 10 machine and one running Windows 7 to see if that makes any difference.

holmesgithub, 2020-04-13:
Hi Droggelbecher42, can you kindly share how you customize the image (default is test.bin) so the new simulated USB storage plugged and detected in Windows host can be large (I always get ~4M with test.bin) said to 2G.
Thank you so much, I posted an issue request #233 relates to this.