RoganDawes/P4wnP1_aloa #130: error in usb coonection

z0m31en7, 2019-06-01:
USB Device not recognized, tried flashing multiple times, tried different cables....

whackyhack, 2019-06-02:
I just learned that p4wnp1_aloa's USB gadgets only works on Pi Zero W, not on Pi Zero.  Could this be the problem?

z0m31en7, 2019-06-02:
> I just learned that p4wnp1_aloa's USB gadgets only works on Pi Zero W, not on Pi Zero. Could this be the problem?

but i am using pi zero w.....

AdNine87, 2019-06-04:
Same problem and the hotspot is not appearing

whackyhack, 2019-06-05:
@z0m31en7 is WiFi AP working for you?  If yes, ssh in and post dmesg entries related to USB might be useful.

michkz, 2019-06-05:
I have this same issue, however the AP spawns without a problem. My dmesg related to usb looks like this:

root@kali:~# dmesg | grep usb
[    0.130239] usbcore: registered new interface driver usbfs
[    0.130369] usbcore: registered new interface driver hub
[    0.130600] usbcore: registered new device driver usb
[    0.360652] usbcore: registered new interface driver smsc95xx
[    0.366899] usbcore: registered new interface driver usb-storage
[    0.424630] usbcore: registered new interface driver usbhid
[    0.426667] usbhid: USB HID core driver
[    3.427845] dwc2 20980000.usb: 20980000.usb supply vusb_d not found, using dummy regulator
[    3.431693] dwc2 20980000.usb: 20980000.usb supply vusb_a not found, using dummy regulator
[    3.923034] dwc2 20980000.usb: EPs: 8, dedicated fifos, 4080 entries in SPRAM
[    3.925981] dwc2 20980000.usb: DWC OTG Controller
[    3.928037] dwc2 20980000.usb: new USB bus registered, assigned bus number 1
[    3.930137] dwc2 20980000.usb: irq 33, io mem 0x20980000
[    3.983240] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    3.985424] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    3.987446] usb usb1: Product: DWC OTG Controller
[    3.989381] usb usb1: Manufacturer: Linux 4.14.80-Re4son+ dwc2_hsotg
[    3.991281] usb usb1: SerialNumber: 20980000.usb
[    7.547016] usbcore: registered new interface driver brcmfmac
[   13.527144] usb0: HOST MAC 42:63:65:12:34:56
[   13.534547] usb0: MAC 42:63:65:56:34:12
[   13.537857] usb1: HOST MAC 42:63:66:12:34:56
[   13.540390] usb1: MAC 42:63:66:56:34:12
[   13.555820] dwc2 20980000.usb: bound driver configfs-gadget
[   15.378923] usbeth: port 1(usb0) entered blocking state
[   15.379116] usbeth: port 1(usb0) entered disabled state
[   15.379840] device usb0 entered promiscuous mode
[   15.395245] IPv6: ADDRCONF(NETDEV_UP): usb0: link is not ready
[   15.419585] usbeth: port 2(usb1) entered blocking state
[   15.419775] usbeth: port 2(usb1) entered disabled state
[   15.425669] device usb1 entered promiscuous mode
[   15.445149] IPv6: ADDRCONF(NETDEV_UP): usb1: link is not ready
[   15.464390] IPv6: ADDRCONF(NETDEV_UP): usbeth: link is not ready

I can see that it raises a few errors on the usb0 and usb1 interfaces, but I have no clue how I should approach this..

z0m31en7, 2019-06-05:
@whackyhack AP is working nicely, I manually installed p4wnp1 to the kali arm image and it is almost working (sometimes hid attacks don't work).

closed 2019-06-05 by z0m31en7

whackyhack, 2019-06-06:
@michkz Is your target Windows?  Some Windows machines complains about unrecognizable device and @mame82 has offered a solution in #125.  Anyway, you may want to open a separate issue as this one is closed.

whackyhack, 2019-06-06:
> may want to open a separate issue as this one is closed.
I see that you opened #134 and have turned off CDC ECM.  It may help want to transfer diagnostics there.

michkz, 2019-06-06:
Will do @whackyhack thanks!