RoganDawes/P4wnP1_aloa #131: Problems to deploy USB serial while connected to USB Ethernet

whackyhack, 2019-06-02:
This is regarding the Web interface.  If I select serial in the default setting, then click "Deploy", the browser briefly returns a red bar at top showing an "Unknown" error, then returns back to the default setting without serial gadget.  In this setting, I must deselect some devices before a deployment can succeed.  I tried to deselect the two HID gadgets, then serial deployment was successful.

If I power the device via PWR while OTG is connected, the result is the same, whether the browser connects via the USB URL or WiFi URL.

However, if I detach OTG and connect via the WiFi URL, Serial selection will remain despite that brief red bar error message. (I haven't figured out where to look for log files.)

Booting without OTG gives me a different problem which I will report separately: OTG gadgets will not be recognized.

whackyhack, 2019-06-09:
This could be a really simple counting error that prevents more than 4 OTG gadgets can be active simultaneously.  The exact same error is reported when deploying storage gadgets, too. (#74, #137)

mame82, 2019-06-09:
TL;TR try to disable other gadgets.

The USB device controller of Pi0 (DesignWare dwc2) provides 8 USB endpoints. Different gadgets consume a different number of endpoints. An USB RNDIS device, f.e. consumes 3 EPs.
A typical HID device consumes 2 EPs, where one of them is the control EP (EP0) and another one is used for HID report exchange. Now for multiple HID devices, EP0 is shared, but each needs a dedicated second EP at minimum. In other words, it is not easy to count the number of EPs needed by a specific gadget setup. ALOA tries anyways and presents an error for most setups which consume more EPs than available. Unfortunately Linux provides no easy way (at least I don't know one) to count consumed EPs **before** deploying the gadget settings. If the P4wnP1 logic fails to guess the correct number and tries to deploy the gadget, even if too many EPs are consumed, the USB stack will simply crash (kernel driver). USB won't work till Pi is rebooted (not possible to recover from such a state, without unloading all relevant kernel modules; even this wouldn't work in all cases).

Best option is to enable needed gadgets one by one and deploy everytime, till the requirements are met. If P4wnP1 crashes, revert to last working setup.

The only alternative I currently see is to only allow specific gadget combinations, which are known to work. This would reduce flexibility.

Worth mentioning: Having USB RNDIS and CDC ECM enabled at the same time consumes most of the EPs available. In most cases, only one of these USB ethernet gadgets is needed (RNDIS for Windows, most other OS'es support CDC ECM)

mame82, 2019-06-09:
Additional note:
Reconfiguration of USB gadgets is best done via WiFi. This is because all deployed gadgets have to be disabled to change if a single parameter and the whole USB device has to be built up from scratch, again (internally). This again means, all USB based network interfaces are re-created, which again means, all depending services/configurations have to be brought up again (interface IP configuration, DHCP etc). All pending network connections would die (especially external tools like nectar, metasploit etc) could end up in unknown state, if bound to an USB ethernet interface. Same is true for the webclient, which relies on persistent TCP sockets (websockets).

I invested several months to hide this from the end user and allow P4wnP1 + its webclient to recover from such a state seamlessly... anyways, that's nearly impossible, as the dependency chain is too complex. In case you want to assure to receive meaningful error messages from the webclient, while re-configuring USB, please use a WiFi connection.

Cheers

mame82, 2019-06-09:
One more note:

If P4wnP1 fails to deploy a gadget configuration and is able to recover (no kernel crash), it will redeploy the last configuration known to work.

The webclient internally receives a configuration update. Once this update is visualized, it looks like nothing has changed (which is true, but there was an error).

To get proper error messages, follow the advice given above.

mame82, 2019-06-09:
@wayhack

Thx for helping other users with issues. I took note of this, even if I'm often short on time to answer questions myself. 

This is how the community should work!

whackyhack, 2019-06-10:
I am more than glad to be helpful, not to say thankful for your projects!  I also appreciate the technical insights you share in comments.  In this particular issue, I totally agree that offering "valid" combination would be a step backward.  Maybe the UI (with JavaScript) could do a precalculation for the end user, disable "Deploy" action when the action is bound to fail. (That Web client via USB would get disconnected when redeploying doesn't bother me at all.)

closed 2019-06-10 by whackyhack

poritz, 2019-06-12:
Hi, I had a similar problem and I think my answer could help others:

I was able to deploy any constalation with RNIDS and stuff or with Massstorage and stuff, but not both.
Answer was: I had VID and PID for an actual usb device for usb gadget and the default for RNDIS
taking out the PID (so only VID given) made it possible for me to deploy everything at once.