RoganDawes/P4wnP1_aloa #133: Typo: No connection dialogue says "Not connection to server"

SEWsam, 2019-06-02:
Typo. Dialogue show when connection to server is lost says "Not connection to server" instead of "No connection to server".

BohdanOpyr, 2019-07-10:
Also, HIDScript job details says "Sate", not "State"

21weberer, 2019-09-28:
Be glad you have a working image which can literally control computers from remote areas.

SEWsam, 2019-09-29:
> Be glad you have a working image which can literally control computers from remote areas.

So....? This forum is for issues. Who cares if this issue doesn't affect the usage of the program, I'm reporting an issue whether it is important or not. Also, there is a perfectly working image on the releases page, so not like I'm lucky.

21weberer, 2019-10-08:
i know that. im saying that you should be glad that he did all of the installation and pre-configuration for you. perhaps i was a little harsh though, so my apologies.

closed 2020-11-08 by SEWsam