RoganDawes/P4wnP1_aloa #134: [HELP] Nothing happens when executing HID scripts

michkz, 2019-06-04:
So first of all I've tried to read all the tickets to see if there was a solution somewhere, forgive me if I missed one. I've successfully installed the P4wnP1-ALOA version on my rpi zero w and when I plug it in my windows pc I am able to connect to the AP that spawns. My problem is when I'm trying to execute one of the standard HID scripts nothing happens. I've turned the CDC ECM toggle off and changed the PID and MID. Is there something I'm not seeing here? I hope someone can help me out. If you need any information or something just ask and I'll try to provide as much as I can!

mame82, 2019-06-06:
Which image are you using?

The release image from this repo or from offensive security?

michkz, 2019-06-06:
The one from this repo. v0.1.0-alpha2 to be exact

michkz, 2019-06-06:
I've posted this in a (now) closed issue. But here is my dmesg output for the usb

root@kali:~# dmesg | grep usb
[ 0.130239] usbcore: registered new interface driver usbfs
[ 0.130369] usbcore: registered new interface driver hub
[ 0.130600] usbcore: registered new device driver usb
[ 0.360652] usbcore: registered new interface driver smsc95xx
[ 0.366899] usbcore: registered new interface driver usb-storage
[ 0.424630] usbcore: registered new interface driver usbhid
[ 0.426667] usbhid: USB HID core driver
[ 3.427845] dwc2 20980000.usb: 20980000.usb supply vusb_d not found, using dummy regulator
[ 3.431693] dwc2 20980000.usb: 20980000.usb supply vusb_a not found, using dummy regulator
[ 3.923034] dwc2 20980000.usb: EPs: 8, dedicated fifos, 4080 entries in SPRAM
[ 3.925981] dwc2 20980000.usb: DWC OTG Controller
[ 3.928037] dwc2 20980000.usb: new USB bus registered, assigned bus number 1
[ 3.930137] dwc2 20980000.usb: irq 33, io mem 0x20980000
[ 3.983240] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[ 3.985424] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[ 3.987446] usb usb1: Product: DWC OTG Controller
[ 3.989381] usb usb1: Manufacturer: Linux 4.14.80-Re4son+ dwc2_hsotg
[ 3.991281] usb usb1: SerialNumber: 20980000.usb
[ 7.547016] usbcore: registered new interface driver brcmfmac
[ 13.527144] usb0: HOST MAC 42:63:65:12:34:56
[ 13.534547] usb0: MAC 42:63:65:56:34:12
[ 13.537857] usb1: HOST MAC 42:63:66:12:34:56
[ 13.540390] usb1: MAC 42:63:66:56:34:12
[ 13.555820] dwc2 20980000.usb: bound driver configfs-gadget
[ 15.378923] usbeth: port 1(usb0) entered blocking state
[ 15.379116] usbeth: port 1(usb0) entered disabled state
[ 15.379840] device usb0 entered promiscuous mode
[ 15.395245] IPv6: ADDRCONF(NETDEV_UP): usb0: link is not ready
[ 15.419585] usbeth: port 2(usb1) entered blocking state
[ 15.419775] usbeth: port 2(usb1) entered disabled state
[ 15.425669] device usb1 entered promiscuous mode
[ 15.445149] IPv6: ADDRCONF(NETDEV_UP): usb1: link is not ready
[ 15.464390] IPv6: ADDRCONF(NETDEV_UP): usbeth: link is not ready

If you need any info, let me know

mame82, 2019-06-09:
Can you post output of

`journalctl -u P4wnP1.service`