RoganDawes/P4wnP1_aloa #135: Caps Lock Problem

ryanpoh, 2019-06-05:
For some reason, when the script runs, it runs with "CAPS" button pressed. Which cause words like "WindowsBase" to be "wINDOWSbASE" which causes errors when running the HID.

I have tried just adding press("CAPS") right after powershell opens with (GUI + r), however the solution isn't stable. It works sometimes, and sometimes it doesn't. How do you recommend I go about this?

mame82, 2019-06-09:
Is CAPS LOCK enabled by the script itself?

Can you post details on the HIDScript you are using.

Which language layout is in use?

ryanpoh, 2019-06-10:
Hi mame82, thanks for the reply! However, it is solved. It was all good after restarting the computer. 

Thanks again!

closed 2019-06-10 by ryanpoh