RoganDawes/P4wnP1_aloa #140: anyway to use this on pi zero non-w?

badboy051, 2019-06-14:


whackyhack, 2019-06-14:
In its current design, not much.  The USB OTG interfaces will not load unless pi0w's built-in WiFi hardware responds. (See technical explanations in #119.)  You can use a WiFi dongle to perform some limited functions, but P4wnP1 is very much USB-centric.

badboy051, 2019-06-14:
how much are these limited functions??

whackyhack, 2019-06-14:
There are a few WiFi attacks. (You can write your own payload, too.)  To see what's available out of the box, just boot up with WiFi dongle, then connect to the Web UI.

ALOA also supports Bluetooth networking so you may try a BT dongle; I am not sure if BT HID is available. (I don't have BT dongle and haven't tested BT on pi0w.)

All in all, Pi0 is more useful with legacy P4wnP1.

badboy051, 2019-06-14:
tnx for info

closed 2019-06-14 by badboy051