RoganDawes/P4wnP1_aloa #141: Double Modifiers.

FreeDurok, 2019-06-15:
Hi and Compliment for your job,
i'm testing P4wnP1, there is an issue with it (or other layouts too).
When i try to write trough HID the curly brackets chars " { "  and " } ",  maybe the hid engine does't recognize double modifiers.
In IT layout curly  brackets  combine "MOD_RIGHT_ALT" and "MOD_RIGHT_SHIFT" (or "MOD_LEFT_SHIFT") + "KEY_RIGHT_BRACE" / "KEY_LEFT_BRACE".
but in my tests I could see that only a modifier is recognized or fails to interpret this combination. I tried modifying and adding to the list the missing modifier in the layout, but nothing to do, any suggestions ?? 

------ From dist/keymaps/it.json
... other codes ...
`
            "¬": [{"Modifiers": ["MOD_RIGHT_ALT"], "Keys": ["KEY_GRAVE"]}],
            "¹": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_1"]}],
            "²": [{"Modifiers": ["MOD_RIGHT_ALT"], "Keys": ["KEY_2"]}],
            "³": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_3"]}],
            "¼": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_4"]}],
            "½": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_5"]}],
            "¬": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_6"]}],
            "{": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_7"]}],
            "[": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_8"]}],
            "]": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_9"]}],
            "}": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_0"]}],
            "`": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_MINUS"]}],
            "~": [{"Modifiers": ["MOD_RIGHT_ALT"], "Keys": ["KEY_EQUAL"]}],
"BACKSPACE": [{"Modifiers": [], "Keys": ["KEY_BACKSPACE"]}],`
.... other codes ...

i try to mod in:

... other codes ...
`
            "¬": [{"Modifiers": ["MOD_RIGHT_ALT"], "Keys": ["KEY_GRAVE"]}],
            "¹": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_1"]}],
            "²": [{"Modifiers": ["MOD_RIGHT_ALT"], "Keys": ["KEY_2"]}],
            "³": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_3"]}],
            "¼": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_4"]}],
            "½": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_5"]}],
            "¬": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_6"]}],
            "{": [{"Modifiers": ["MOD_RIGHT_ALT", "MOD_LEFT_SHIFT"],	"Keys": ["KEY_LEFTBRACE"]}],
            "[": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_8"]}],
            "]": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_9"]}],
            "}": [{"Modifiers": ["MOD_RIGHT_ALT", "MOD_LEFT_SHIFT"],	"Keys": ["KEY_RIGHTBRACE"]}],
            "`": [{"Modifiers": ["MOD_RIGHT_ALT"],	"Keys": ["KEY_MINUS"]}],
            "~": [{"Modifiers": ["MOD_RIGHT_ALT"], "Keys": ["KEY_EQUAL"]}],
"BACKSPACE": [{"Modifiers": [], "Keys": ["KEY_BACKSPACE"]}],`

but not working for me.

thx guys i hope anyone can suggest to me any tips :)

mame82, 2019-06-16:
`"{": [{"Modifiers": ["MOD_RIGHT_ALT", "MOD_LEFT_SHIFT"],	"Keys": ["KEY_LEFTBRACE"]}],` looks correct to me. Have you restarted P4wnP1 before testing?

FreeDurok, 2019-06-16:
mmm... yes i tryed to reboot and not, but today i'll retry, Thx a lot.
anyway to make curly brackets in IT layout this is the right way :)

FreeDurok, 2019-06-18:
I tried right now, but although I made the change and restarted, it didn't work :(

mame82, 2019-07-24:
Note: Further investigation needed, tracking as keyboard layout bug