RoganDawes/P4wnP1_aloa #145: New pi4

PinkPantherX, 2019-06-26:
A new raspberry pi4 is out. Can this be installed on it? Would it work better, has anyone tried it?

TheMMcOfficial, 2019-06-26:
I don't think... It doesn't support the raspberry pi 3/3b+. It's design to be a little stealthy. Maybe if a new raspberry pi zero is released it Mame will adapt the code for it.

whackyhack, 2019-06-26:
Like Pi3 and all previous "proper" Pi's, Pi4 still doesn't support USB gadgets, so is of little use with P4wnP1.

Googulator, 2019-10-18:
Actually the Pi 4 can act as an USB gadget through the USB-C port. The Pi 3A+ also supports gadget mode.

Sp8c3, 2020-11-17:
wouldn't it be nice to use a Pi 4 to add features like a keylogger, or even a LanTap, and maybe you could even do Video-MITM wiht a dongle like [this](https://www.aliexpress.com/item/4001043540669.html) or an Adapter like [this](https://www.aliexpress.com/item/4000102166176.html) :)