RoganDawes/P4wnP1_aloa #147: External adapter issue

bollapinnsvin, 2019-07-01:
Hi
For some days i have been grinding my gears on a usb wifi adapter issue i can't figure out.
I want to use a external Alfa  AWUS036NEH adapter with the P4wnP1 since its more powerful and detects wifi over a far larger area.
But when i connect it, 70% of the time it gets assigned wlan0, so the internal wifi gets disabled.
I really want the adapter to be assigned wlan1 so i can use the internal wifi to control it.
I have tried to create the 70-persistent-net.rules file and tried all kinds of tricks to permanently assign the internal wifi wlan0 with now success, it just shows as rename3 with 70-persistent-net.rules.
Is there som config files that i'm not aware of that i can modify.

BTW I made this case to make the P4wnP1 super useful, usb stick with built in 500mAh battery and charger. Runs the P4wnP for 2h 40m straight https://www.thingiverse.com/thing:3693266

whackyhack, 2019-07-02:
If power-range is the main goal for using the external adapter, you may consider an [external antenna for the build-in adapter](https://www.briandorey.com/post/raspberry-pi-zero-w-external-antenna-mod).

That case is very cool.

bollapinnsvin, 2019-07-02:
Well, the thing is that i want to keep the form factor, and still be able to add on adapters and other stuff if needed. Like SDR and 5 GHz wifi, and still be able to control it with the builtin wifi. I also wanted to make a little case for this wifi adapter https://shop.hak5.org/collections/wifi-gear/products/ralink-usb-wifi-rt5370
so i can use the pi as a really small pineapple and do all kinds of fun stuff :P

mame82, 2019-07-24:
Not supported: Current version of P4wnP1 is aimed to run in USB device mode. Host mode isn't disabled, but I don't give support for this cases (yet)

Dtpk, 2020-05-01:
> Not supported: Current version of P4wnP1 is aimed to run in USB device mode. Host mode isn't disabled, but I don't give support for this cases (yet)

When you do provide support will you make a update? I am interested in using external usb WiFi adapters as well I even got a powered otg hub before I realized it had something to do with the firmware and not my hardware. Besides that everything else software wise been great thank you for making this image it is my goto in my portable I made.

SymbianSyMoh, 2020-05-01:
@Dtpk From within the SSH access of P4wnP1 do the following:
1. `nano /boot/config.txt`.
2. Comment the line that says `dtoverlay=dwc2`.
3. Reboot.
4. Now all USB ports in your P4wnP1's PI board are in HOST mode and you can use whatever external USB devices you want.

Want to go back to GADGET mode, reverse step #2 and you are good to go!