RoganDawes/P4wnP1_aloa #148: Wifi Covert Channel not working on Win8.1 PRO

himuura, 2019-07-02:
Bypassing the original problem with the keyboard layout (added the US keyboard to win8.1 just to test), executing the wifi_covert_channel errors.

Exception calling "Invoke" with "1" argument(s): "Invalid identifier.
Nome do parâmetro: handle"
At line:26 char:8
+ return $c.Invoke($a)
+        ~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException
    + FullyQualifiedErrorId : ArgumentException

stage2 reassembled
You cannot call a method on a null-valued expression.
At line:106 char:1
+ $devfile.Close()
+ ~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (:) [], RuntimeException
    + FullyQualifiedErrorId : InvokeMethodOnNull

You cannot call a method on a null-valued expression.
At line:107 char:1
+ $devfile.Dispose()
+ ~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (:) [], RuntimeException
    + FullyQualifiedErrorId : InvokeMethodOnNull

Invoke-Expression : Cannot bind argument to parameter 'Command' because it is an empty string.
At line:102 char:5
+ iex $stage2
+     ~~~~~~~
    + CategoryInfo          : InvalidData: (:) [Invoke-Expression], ParameterBindingValidationException
    + FullyQualifiedErrorId : ParameterArgumentValidationErrorEmptyStringNotAllowed,Microsoft.PowerShell.Commands.Invo
   keExpressionCommand

Any help would be greatly appreciated!

himuura, 2019-07-02:
The powershell does open and it writes the code but there might be something odd...

mame82, 2019-07-24:
Note: Possible bug

himuura, 2019-07-24:
@mame82 i think it might have something to do with the keyboard layout...i've already tried to build a JSON file with my keyboard but with no avail...the Z and Y key are being switched, dunno why. I've already opened an issue here. Could you check it out, please?