RoganDawes/P4wnP1_aloa #149: is there a way to do different preset things on the PI easily, e.g. shorting two pins?

r0tati0n, 2019-07-04:
Hello all,

is there a way to do different preset things on the PI easily, e.g. shorting two pins?

For example:
- Shorting pins 1-2 executes a HIDscript when plugged in
- Shorting pins 3-4 executes a different HIDscript
- Shorting pins 5-6 activates WIFI
- Shorting pins 7-8 boots up with USB ethernet
- Shorting pins 9-10 activates bluetooth
- Shorting pins 5-6 and 9-10 activates both WIFI and bluetooth

mame82, 2019-07-24:
Yes and no.

There is a GPIO based Trigger, which could detect state changes on GPIO pins.

It partially fulfills your requirement. Issue is, that you can't detect current PIN state on demand (on boot in your case), as this is no event.

A possible solution would be a bash script, which triggers "on service start" (boot).
This script could read GPIO pin states and take a decision on how to progress.
To give control back to P4wnP1, for each branch in this bash script `P4wnP1_cli` could be called to send a Trigger to a dedicated group channel (channel name depending on branch, f.e. "PIN1_HIGH")

Now the respective group channel triggers could be configured to your needs in the TriggerAction tab of the web-ui

mame82, 2019-07-24:
Okay, carefully re-reading your request, I assume you are thinking about some kind of boot-up jumper configuration (pin 1-2 shortened, pi 3-4 shortened ...).

This brings me to an idea, which could solve your request, but needs to be tested.

Step 1) Bring Pin 1,3,5,7... to a defined state on boot (let's say HIGH f.e.). This could be achieved with respective TriggerActions - using trigger "on service start" and GPIO output as Action.

Step 2) Poll Pin 2,4,6,8 ... for a state change to high. Assuming these PINs are low in default state and high when shortened by a jumper. F.e. shortening 1-2 would bring Pin 2 to high, because 1 is set to high in step 1. To achieve this, again, TriggerActions could be used, with Trigger "GPIO input" (which also could be used to define the default state to low) and the Action you want to run, when the respective PIN is shortened.

As stated, this idea needs testing.