RoganDawes/P4wnP1_aloa #150: HID Keyboard Drivers missing?

NotTimTam, 2019-07-10:
Installed from Kali.org to RasPiZeroW set up some basic scripts and got stuff running. Tried it on a windows PC, windows installed the drivers and mouse inputs work. Also typing text in works. But doing things like
`press("GUI r")`
Doesn't work?! Also if I go to devices in the new windows ten settings app, it shows my P4wnP1 devices name, but says `drivers unavailable` underneath!

NotTimTam, 2019-07-10:
Update: Running the included 'helper.js' script works. But nothing else

NotTimTam, 2019-07-10:
UPDATE: Keyboard inputs work, I found an undetected error in the included script...

ONLYA, 2019-07-10:
Me too. I have tried but it seems that the hid is not working even if there is /dev/hidg1 existing.

NotTimTam, 2019-07-10:
IT seems I got it working. But it types 'y's as z and types apostrophes as |

RoganDawes, 2019-07-11:
Sounds like you have the wrong keyboard mapping selected.

NotTimTam, 2019-07-11:
What should I use, I thought I was using the one for the US.

RoganDawes, 2019-07-11:
Not sure. Perhaps it is currently set to DE, since Marcus is German.

Think i read something about that in the docs today.

NotTimTam, 2019-07-11:
🤦‍♂ duh.

closed 2019-07-11 by NotTimTam