RoganDawes/P4wnP1_aloa #151: This doesn't work on my computer at all

ONLYA, 2019-07-10:
I have flashed the image many times. Actually, the network works as I am able to adjust the wireless settings without a problem. However, everything about the USB connection gets all wrong.
At first, it is not recognized as a USB HID device and hence the HID emulation cannot be completed. However, the feature works because it can successfully finish running `console.log("HI")` individually. But it just keeps running for ages with HID commands. (I have tried with manual HID emulation by directly execute `echo -ne "..." > /dev/hidg0` and it also kept stuck without any results.)
Secondly, I cannot use mass storage function yet. All are just fine. But after I click Deploy, nothing happens.
Thirdly, I cannot change the system password. It seems that the system has been pre-set the auto auth for "root-toor" only because I can never find it in my network after rebooting.
Finally, I think you are making a USB-serial gadget so it should be recognized as a COM-PORT after I set it as serial through USB. But it didn't.

My OS is Windows 10. I was connecting the Pi 0 w on a USB hub, but it doesn't work for the rest of USB ports. I am sure that the ports work because I have used them for development through COM communication before.
Could someone offer me a suggestion or something to learn? Thanks.

dopescope45, 2019-07-21:
it says i need a driver to  use the keyboard and i have yet to find one

21weberer, 2019-09-18:
@ONLYA which method are you using? are you using a usb cable or did you solder on a usb head? the usb female ports may be fine, but the cable you are using might not be. try different cables, as I had the same HID problems until i replaced my USB head. To change the system password, ssh into the pi with root@172.24.0.1 when connected to the Access point. then, use psswd to change the password. @colin773, try changing the VID, PID, SN, etc. in the usb gadget settings page.

ONLYA, 2019-09-29:
Just a USB cable that is connected to the DATA port. Have changed various cables including the cable given by Texas Instruments for Emulator usage. That just didn't work... Kinda of thinking whether my computer cannot recognize the COM port properly...

21weberer, 2019-10-10:
I wonder, can you take a picture of your USB settings page? maybe something is incorrectly configured there.

ONLYA, 2019-10-14:
Sorry, what do you mean by USB settings page? Device Manager?

21weberer, 2019-10-14:
USB settings, as seen here:
[image]