RoganDawes/P4wnP1_aloa #152: Powershell on covert channel: "Closing handle to native WiFi API"

b1116490, 2019-07-12:
Hello. When I try to use the WiFi covert channel script on my own PC, I get this error after stage 2 presumably starts: 

Process died with exit code: -3
Closing handle to native WiFi API
Process died with exit code: -3
Closing handle to native WiFi API
Process died with exit code: -3
Closing handle to native WiFi API
Process died with exit code: -3
Closing handle to native WiFi API
Process died with exit code: -3
Closing handle to native WiFi API
Process died with exit code: -3
Closing handle to native WiFi API
Process died with exit code: -3
Closing handle to native WiFi API
Process died with exit code: -3

etc. Not sure what more details I should add or how to troubleshoot.

b1116490, 2019-07-19:
I've seen tried on another PC and seemingly got further but with no successful remote control. Once the stager is ran, the powershell windows continuously says 
"==scan ==
Network Available! SSID recieved: [ssid of local wifi ap] 
Discarded SSI: [same SSID] 
SSID recieved: [different ssid] 
etc 
==scan end==.. To be honest I don't understand why this needs to 

On the P4wnP1, I sometimes get to the message "InRsp2..." with the session listed but nothing happens when I connect to it, and sometimes it just stops after InRsp1.

b1116490, 2019-07-19:
Apologies, I seem to have made progress through random trial-and-error, but now I am at a roadblock again. My victim PC doesn't seem to want to connect to the hidden AP, which I only saw for a brief moment in the list of WiFi spots, but I suppose that's what makes it hidden. Trying to connect to that manually fails. I am at a loss.

mame82, 2019-07-24:
What exactly have you done? Ran the bash script or deployed the covert channel master template?

There is no hidden AP to connect to, the covert channel hides data in 802.11 probe requests/responses

b1116490, 2019-07-29:
Sorry, I did say in the first post that I deployed the covert channel master template, but I should have been more clear. 
I have also assumed a few things about how it works, from the behavior of my PC when I launch the attack and the things I read on the wiki.

Here is precisely what I do:

1. Plug in the P4wnP1 with nearly default settings, with only the WiFi and BT access point name and passwords changed.
2. Connect to it through BT, and deploy the covert channel master template.
3. Let it enter the payload into powershell, with an US keyboard, no issues there.
4. This shows up on the PS window after the payload is entered: 

> Start receiving stage2
..................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................stage2 reassembled

>GAC    Version        Location
>---    -------        --------
>False  v2.0.50727
>Starting test
>Trying to connect covert channel via '802.11n USB Wireless LAN Card'
>Connect Init Request 1 (to srvID 9)...
>Scan completed!
>Network Available!
>SSID recieved: (censored)
>Discarded SSID: (censored)
>SSID recieved: (censored)
>Discarded SSID: (censored)
>Scan completed!
>SSID recieved: (censored)
>Discarded SSID: (censored)
>SSID recieved: (censored)
>Discarded SSID: >(censored)
>Network Available!
>Scan completed!
>SSID recieved: (censored)

And so on. From there I suspect something might be wrong, since this isn't what it looked like on videos showing the process.

5. SSH into the P4wnP1. do screen -d -r wifi_c2. The arguments might be the other way around
6.This is what I see on the terminal next:

> Bound to server ID 9
>Listening for incoming connections (max 15)
>MaMe82 WiFi covert channel > InReq1: Connection request from client IV: (some code)
>...InRsp1: Handing out client ID 1

From there on I have access to commands. When I type sessions however I get no results, and when I type interact 1 it tells me that I need to enter a valid client ID according to the sessions command.
Note that the PS window is still looping through the same messages.

The odd thing is that, while following the same procedure before, I have gotten different results. Sometimes I would get a result when typing sessions, but with no success when trying to interact with it. I have also had different results on PS.

When I exit out of screen on ssh, I can see some errors from /usr/local/P4wnP1/legacy/wifi_server.py, namely in order lines 1189, 1045 and 747.

Thank you for your reply.

MasterX135, 2019-09-27:
Hello. I have the same problem mentioned by b1116490

DarkTech2, 2019-12-14:
I'm running into the same issues here -- adding some additional information:

I manually SSH into the RPi0w and run the wifi_covert_channel.sh before deploying the hidscript. All other reqs are in place (usb gadget options from the template)

The payload is deployed fine and leaves the same display on the PS window of receiving and discarding SSIDs on loop.

The SSH window that ran wifi_covert_channel.sh shows "Ended stage2 delivery", but no other prompts, etc. No other commands can be typed here without interrupting the script.

The screen -d -r wifi_c2 shows "InReq1: Connection request from client IV: [code]" and "InRsp1: Handing out client ID 1", but doesn't show any response or the addition of that IV into the accept-queue like the seytonics video does. Sessions lists nothing. You can't use interact to interact with any since there aren't any sessions available.

I've reimaged the SD card, and tested on 3 different machines with different specs/HW/etc running Windows 10 Build 1903 on each. Same results.

Screenshots of screen command, the sh script, and the payload on victim.

[image]
[image]
[image]

Franci01, 2019-12-16:
same exact issue here, please let me know if you find any solution

wthorstad, 2019-12-19:
I am also experiencing this issue, wondering if anyone has figured it out?

newberryb, 2020-01-07:
I am having the same issue getting 
Process died with exit code: -1
Closing handle to native WiFi API
Process died with exit code: -1
Closing handle to native WiFi API
Process died with exit code: -1
Closing handle to native WiFi API
Process died with exit code: -1
Closing handle to native WiFi API
Process died with exit code: -1
Closing handle to native WiFi API
Process died with exit code: -1
Closing handle to native WiFi API
Process died with exit code: -1
Closing handle to native WiFi API
Process died with exit code: -1
Did Microsoft patch the WIFI API to fix this vulnerability?

beerinfinity, 2020-11-19:
Exactly the same issue encountered - anyone found a resolution as I have spent 2+ days on this one?