RoganDawes/P4wnP1_aloa #155: Incorrect keyboard layout

himuura, 2019-07-18:
Well, i've tried to map my own keyboard without success. For some reason, my best effort resulted in an almost perfect combination, only the Z and Y being switched (despite correct mapping the keys). I think it has something to do with an extra key somewhere that may be causing a misalignment. Anyone knows how to solve this one? Any other config files that i can edit to correct this? Keyboard layout PT-PT UTF-8.
Sample keys in order (in case some kind soul wants to help):

Esc F1 F2 F3 F4 F5 F6 F7 F8 F9 F10 F11 F12 Pause/Break Prt SCR/SysRQ Delete

\ 1 2 3 4 5 6 7 8 9 0 ' «

(Shift) | ! " # $ % & / ( ) = ? »

(alt gr) @ (KEY_2) £ (KEY_3) § (KEY_4) € (KEY_5) { (KEY_7) [ (KEY_8) ] (KEY_9) } (KEY_0) 

(TAB) q w e r t y u i o p + ´

(shift) Q W E R T Y U I O P * `

(alt gr) € (KEY_E)

(CapsLOCK) a s d f g h j k l ç º ~

(shift) ASDFGHJKLÇª^

(SHFT KEY) < z x c v b n m , . -

(shift) > Z X C V B N M ; : _

(the last row i think is the same, CTRL, FN, LFT GUI, ALT, SPACE, ALT GR, MOUSE CLICK, RIGHT CTRL)

himuura, 2019-07-18:
{
        "Name": "PT",
        "Description": "Portuguese USB Keyboard report mapping",
        "Mapping": {

            "CTRL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "CONTROL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "LEFT_CTRL": [{"Modifiers": ["MOD_LEFT_CONTROL"], "Keys": []}],
            "RIGHT_CTRL": [{"Modifiers": ["MOD_RIGHT_CONTROL"], "Keys": []}],
            "ALT": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": []}],
            "LEFT_ALT": [{"Modifiers": ["MOD_LEFT_ALT"], "Keys": []}],
            "RIGHT_ALT": [{"Modifiers": ["MOD_RIGHT_ALT"], "Keys": []}],
            "SHIFT": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": []}],
            "LEFT_SHIFT": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": []}],
            "RIGHT_SHIFT": [{"Modifiers": ["MOD_RIGHT_SHIFT"], "Keys": []}],
            "GUI": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "WIN": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "LEFT_GUI": [{"Modifiers": ["MOD_LEFT_GUI"], "Keys": []}],
            "RIGHT_GUI": [{"Modifiers": ["MOD_RIGHT_GUI"], "Keys": []}],

            "ESC": [{"Modifiers": [], "Keys": ["KEY_ESC"]}],
            "ESCAPE": [{"Modifiers": [], "Keys": ["KEY_ESC"]}],

            "F1": [{"Modifiers": [], "Keys": ["KEY_F1"]}],
            "F2": [{"Modifiers": [], "Keys": ["KEY_F2"]}],
            "F3": [{"Modifiers": [], "Keys": ["KEY_F3"]}],
            "F4": [{"Modifiers": [], "Keys": ["KEY_F4"]}],
            "F5": [{"Modifiers": [], "Keys": ["KEY_F5"]}],
            "F6": [{"Modifiers": [], "Keys": ["KEY_F6"]}],
            "F7": [{"Modifiers": [], "Keys": ["KEY_F7"]}],
            "F8": [{"Modifiers": [], "Keys": ["KEY_F8"]}],
            "F9": [{"Modifiers": [], "Keys": ["KEY_F9"]}],
            "F10": [{"Modifiers": [], "Keys": ["KEY_F10"]}],
            "F11": [{"Modifiers": [], "Keys": ["KEY_F11"]}],
            "F12": [{"Modifiers": [], "Keys": ["KEY_F12"]}],
            "SYSRQ": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "PRINT": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "PRINTSCR": [{"Modifiers": [], "Keys": ["KEY_SYSRQ"]}],
            "SCROLLLOCK": [{"Modifiers": [], "Keys": ["KEY_SCROLLLOCK"]}],
            "SCROLL": [{"Modifiers": [], "Keys": ["KEY_SCROLLLOCK"]}],
            "PAUSE": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],
            "BREAK": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],
            "BRK": [{"Modifiers": [], "Keys": ["KEY_PAUSE"]}],

            "\": [{"Modifiers": [], "Keys": ["KEY_GRAVE"]}],
            "1": [{"Modifiers": [],     "Keys": ["KEY_1"]}],
            "2": [{"Modifiers": [],     "Keys": ["KEY_2"]}],
            "3": [{"Modifiers": [],     "Keys": ["KEY_3"]}],
            "4": [{"Modifiers": [],     "Keys": ["KEY_4"]}],
            "5": [{"Modifiers": [],     "Keys": ["KEY_5"]}],
            "6": [{"Modifiers": [],     "Keys": ["KEY_6"]}],
            "7": [{"Modifiers": [],     "Keys": ["KEY_7"]}],
            "8": [{"Modifiers": [],     "Keys": ["KEY_8"]}],
            "9": [{"Modifiers": [],     "Keys": ["KEY_9"]}],
            "0": [{"Modifiers": [],     "Keys": ["KEY_0"]}],
            "'": [{"Modifiers": [],     "Keys": ["KEY_MINUS"]}],
            "«": [{"Modifiers": [], "Keys": ["KEY_EQUAL"]}],

            "|": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_GRAVE"]}],
            "!": [{"Modifiers": ["MOD_LEFT_SHIFT"],     "Keys": ["KEY_1"]}],
            "\"": [{"Modifiers": ["MOD_LEFT_SHIFT"], "Keys": ["KEY_2"]}],
            "#": [{"Modifiers": ["MOD_LEFT_SHIFT"],     "Keys": ["KEY_3"]}],
            "$": [{"Modifiers": ["MOD_LEFT_SHIFT"],     "Keys": ["KEY_4"]}],
            "%": [{"Modifiers": ["MOD_LEFT_SHIFT"],     "Keys": ["KEY_5"]}],

            "&": [{"Modifiers": ["MOD_LEFT_SHIFT"],     "Keys": ["KEY_6"]}],
            "/": [{"Modifiers": ["MOD_LEFT_SHIFT"],     "Keys": ["KEY_7"]}],
            "(": [{"Modifiers": ["MOD_LEFT_SHIFT"],     "Keys": ["KEY_8"]}],
            ")": [{"Modifiers": ["MOD_LEFT_SHIFT"],     "Keys": ["KEY_9"]}],
            "=": [{"Modifiers": ["MOD_LEFT_SHIFT"],     "Keys": ["KEY_0"]}],
[... 130 lines cut ...]
            "UP": [{"Modifiers": [], "Keys": ["KEY_UP"]}],
            "RIGHT_ARROW": [{"Modifiers": [], "Keys": ["KEY_RIGHT"]}],
            "LEFT_ARROW": [{"Modifiers": [], "Keys": ["KEY_LEFT"]}],
            "DOWN_ARROW": [{"Modifiers": [], "Keys": ["KEY_DOWN"]}],
            "UP_ARROW": [{"Modifiers": [], "Keys": ["KEY_UP"]}],

            "COMPOSE": [{"Modifiers": [], "Keys": ["KEY_COMPOSE"]}],

            "NUMLOCK": [{"Modifiers": [], "Keys": ["KEY_NUMLOCK"]}],
            "NUM": [{"Modifiers": [], "Keys": ["KEY_NUMLOCK"]}]

           // "F13": [{"Modifiers": [], "Keys": ["KEY_F13"]}],
            //"F14": [{"Modifiers": [], "Keys": ["KEY_F14"]}],
            //"F15": [{"Modifiers": [], "Keys": ["KEY_F15"]}],
            //"F16": [{"Modifiers": [], "Keys": ["KEY_F16"]}],
            //"F17": [{"Modifiers": [], "Keys": ["KEY_F17"]}],
            //"F18": [{"Modifiers": [], "Keys": ["KEY_F18"]}],
            //"F19": [{"Modifiers": [], "Keys": ["KEY_F19"]}],
            //"F20": [{"Modifiers": [], "Keys": ["KEY_F20"]}],
            //"F21": [{"Modifiers": [], "Keys": ["KEY_F21"]}],
            //"F22": [{"Modifiers": [], "Keys": ["KEY_F22"]}],
            //"F23": [{"Modifiers": [], "Keys": ["KEY_F23"]}],
            //"F24": [{"Modifiers": [], "Keys": ["KEY_F24"]}]
    }
}

himuura, 2019-09-04:
So no keyboard support other than US right?

Blasttt, 2019-12-20:
I'm after the same thing :/

carolinalopes98, 2020-01-01:
@himuura hi, this is not related to your question, im sorry, i couldnt solve the messing y and z, but im having trouble adding the json file to the keymap folder because when I go to look at the files present in the folder in question in ssh via putty the file already present inside its green while my pt.json file is written in white. can you help me with this pls?

PineMaster, 2020-03-26:
umm hello use single quotes instead of double quotes @himuura