RoganDawes/P4wnP1_aloa #158: [Question] How do i connect P4wnP1 aloa to wifi?

SkyCZ, 2019-07-27:
How can i connect the device to wifi so i can clone from github and do other stuff?

himuura, 2019-07-29:
wifi settings, in there you have a drop down list currently set to AP. Change it to Client with Failover to AP and set the WiFi of your network accordingly (check this tutorial: https://jamesachambers.com/kali-linux-p4wnp1-aloa-guide-setup-usage-examples/ ). I usually do it while connected to the microUSB PWR instead of data, just to be safe and dont inject something by mistake.

testbughub, 2019-08-20:
I'm not trying to hijack or anything, but I've gor an issue with connecting to an AP.
I've been trying to get it to connect to my phone's hotspot, but it keeps stalling on "acquiring IP" or whatever. 
It just get stuck there and never get's it's IP.
How do I troubleshoot? UART? Or is there a better way?